Azure/iot-edge-opc-publisher-v1 #1: Cannot load JSON configuration file

RavilN, 2016-11-23:
Hello,
I am trying to build and run the IoT Gateway with OPC UA Client module, and have issues with loading of JSON configuration file. The code expects that the node for name of module in JSON is "name",  while in file it is defined "module name". If I correct that to match, later it fails in function gateway_addmodule_internal due to loading args being not defined (module_entry->module_loader_info.entrypoint).
And the file dotnet.dll cannot be found.
I wonder is this up to date version? Can you get it from source code to clean machine and run without errors?

Thank you, Ravil

606simps, 2016-11-24:
This OPC-UA Client module is targeted for IoT Gateway 2016-11-02 for now. The latest IoT Gateway is 2016-11-18, and has changed some data structure. You will need to use IoT Gateway 2016-11-02 to get it work.

RavilN, 2016-11-24:
I figured out that dotnet binding is not enabled by default, should be enabled before building of the IoT Gateway SDK. Modified json file looking at samples in IoT Gateway, and then was able to load OPC UA Client module. Arguments to it could not be parsed, so for now hard coded device Id and shared key.
And endpoints could not be read from config file, for now just hard coded too.
After these manipulations I was able to push data to the IoT Hub.
When you synchronize code for UA Client with latest IoT Gateway, then it should work I guess.
BTW, if you need some demo servers to connect to, accessible from the Internet, there are some at opcuaserver.com.

barnstee, 2016-11-25:
Code is now sychonised with first official release 2016-11-18, please try it out. Loading config files and passing args is also working for me. Building the gateway with dotnet binding is explicitly called out in bold, this is important.

RavilN, 2016-11-25:
I tried to clone both projects and rebuild them, from master branch. The IoT Gateway SDK project now cannot be built: CMake error at testtools/micromock/CMakeLists.txt:60 (file):
 file FILE([TO_CMAKE_PATH | TO_NATIVE_PATH\ path result) must be called with exactly three arguments.
Also tried to call it with option --skip-unittests, hoping that it would not try to build test projects, the same error.
Then tried to build newest OPC UA Client Module with the binaries of the version of IoT Gateway I had already (a few days old one), it builds, but cannot run, throws runtime error. Perhaps there is some mismatch.
For now going to stay with my local modified version which works after some minor changes.

barnstee, 2016-11-28:
What's the runtime error you're getting?

606simps, 2016-11-28:
Mine issue is solved. It was a typo of ***SecurityMode*** in the ***Opc.Ua.Client.SampleModule.Endpoints.xml***. However I noticed one thing described below.

The arguments passed to the module is actually including double quotes. So with original parsing code, the deviceid and password are both not correct.
```C#
m_DeviceID = configurationString.Substring(0, configurationString.IndexOf(';'));
m_SharedAccessKey = configurationString.Substring(configurationString.IndexOf(';') + 1);
```
The deviceid will be prefixed with a " and the key will be suffixed with a ". I suppose this is a Gateway bug.

(I used 2016-11-18 Gateway SDK and 2016-11-25 OPC UA client)

Thanks for your great work and help!
Stephen

damonbarry, 2016-11-28:
@RavilN I'm on the gateway SDK team...I'm curious about the CMake error in testtools/micromock/CMakeLists.txt:60. What's the value of your VCInstallDir environment variable?

Also, what steps did you take to build the SDK? Did you use tools\build.cmd, or direct CMake commands?

I'm not able to repo this problem, so I'd love to figure out what's different between our environments...

RavilN, 2016-11-29:
Looks like the runtime error was caused by version mismatch. I got latest version of the IoT Gateway SDK and OPC UA Client Module (Nov 28), rebuilt all and now it runs.
The IoT Gateway SDK did not build Microsoft.Azure.IoT.Gateway.dll although the option to build dotnet binding was set. I built it from the solution.

Device Id had  quote sign at start and shared key had it at the end, as 606simps noted. I fixed that in my local code, but did not work anyway. Because it was adding back slash ( \ )before forward slash ( / ) in the shared key. Something to do with encoding or reading of config file. When I hard coded the key, it was able to connect to IoT Hub and publish messages successfully.

RavilN, 2016-11-29:
Damon, the VCInstallDir variable was correct.  Now it builds, not sure what was the cause.

barnstee, 2016-11-29:
OK, excellent, I will open issues for these...

closed 2016-11-29 by barnstee

barnstee, 2016-12-02:
This is now fixed. It was actually a problem in the module.