Azure/iot-edge-opc-publisher-v1 #4: Stupid question?

gillerman, 2016-12-14:
When I try to build the binding sample I get the following errors:

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)		cannot open source file "stdio.h"	dotnet_binding_sample	c:\Users\John\Documents\Visual Studio 2015\Projects\iot-gateway-opc-ua\binding\main.c	4	

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)		cannot open source file "stddef.h"	dotnet_binding_sample	c:\Users\John\Documents\Visual Studio 2015\Projects\iot-gateway-opc-ua\inc\azure_c_shared_utility\buffer_.h	12

barnstee, 2016-12-15:
The only stupid question is the one that didn't get asked. :-)
You are right, the dotnet binding project only builds the managed Microsoft.Azure.IoT.Gateway.dll. You need to build the entire SDK (the C libraries) using the build command in tools and don't forget to enable the dot net language binding. Just do a build /? and you'll find it. You also seem to be missing standard C libraries and headers from your VS path. Do you have everything set up correclt yin VS, e.g. does another C++ console app or something like that build OK?

barnstee, 2016-12-19:
@gillerman, is your issue resolved? Then please close...

gillerman, 2016-12-19:
Complete

closed 2016-12-19 by gillerman