Azure/iot-edge-opc-publisher-v1 #12: Error when running the example i.e. creating session

sumongithub, 2017-01-27:
Hi,
Please find below error message when running the example i.e. creating session. 

E:\..\iot-gateway-opc-ua\bin\x86\Debug>dotnet_binding_sample.exe gateway_config.json
Opc.Ua.Client.SampleModule: Creating session for endpoint: opc.tcp://localhost:27830/
Opc.Ua.Client.SampleModule: WARNING: Auto-accepting certificate: CN=UaOmniServer@LenovoPC, OU=Unit, O=Organization, L=LocationName, S=State, C=US, DC=LenovoPC
Opc.Ua.Client.SampleModule: Could not connect to updated endpoint opc.tcp://localhost:27830/. Exception: Opc.Ua.ServiceResultException: Error establishing a connection: Error received from remote host:
   at Opc.Ua.Client.Session.<Create>d__72.MoveNext() in C:\Users\mregen\Source\Repos\UA-.NETStandardLibrary\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 810
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at Opc.Ua.Client.SampleModule.<EndpointConnect>d__6.MoveNext() in E:\..\iot-gateway-opc-ua\Module.cs:line 107
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()
   at Opc.Ua.Client.SampleModule.<Create>d__5.MoveNext() in E:\..\iot-gateway-opc-ua\Module.cs:line 89
Opc.Ua.Client.SampleModule: OPC UA Client Sample Module created.
Opc.Ua.Client.SampleModule: ERROR: Could not find endpoint URL opc.tcp://localhost:27830/ in active server sessions, NodeID 2258 NOT published!
Opc.Ua.Client.SampleModule: To fix this, please update your config.xml with the updated enpoint URL!
Opc.Ua.Client.SampleModule: ERROR: Could not find endpoint URL opc.tcp://localhost:27830/ in active server sessions, NodeID 2258 NOT published!
Opc.Ua.Client.SampleModule: To fix this, please update your config.xml with the updated enpoint URL!
Opc.Ua.Client.SampleModule: OPC UA Client Sample Module started.
gateway successfully created from JSON
gateway will run until ENTER is pressed

barnstee, 2017-01-27:
Looks like the server refused the connection. Cert issue? I recently updated the error messages, are you using the latest from master?

sumongithub, 2017-01-31:
Thank you Erich, Now it is working fine after adding certificate into trusted group. 

**One question:**
How to add custom tag in configuration file, e.g. Namespaceindex=02, IdentifierType=String and Identifier=test02.sumon01

but it is not appearing in the azure message. 

```
<Extensions>
    <ua:XmlElement>
      <ListOfPublishedNodes xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd">
        <NodeLookup>
          <ua:EndpointUrl>opc.tcp://localhost:27730</ua:EndpointUrl>
          <ua:NodeId>
            <!-- Current Server Time -->
            <ua:Identifier>i=2257</ua:Identifier>
          </ua:NodeId>
          <ua:NodeId NamespaceIndex="2">
            <ua:Identifier>s=test02.sumon01</ua:Identifier>
            <!--<ua:BrowseName>2,"test02.sumon01"</ua:BrowseName>-->
          </ua:NodeId>
        </NodeLookup>
      </ListOfPublishedNodes>
    </ua:XmlElement>
  </Extensions>
```

barnstee, 2017-02-01:
Namespaces needs to be specifid as part of the identifier and strings needs to be in quotes, i.e. it should be:
<Extensions>
    <ua:XmlElement>
      <ListOfPublishedNodes xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd">
        <NodeLookup>
          <ua:EndpointUrl>opc.tcp://localhost:27730</ua:EndpointUrl>
          <ua:NodeId>
            
            <ua:Identifier>i=2257</ua:Identifier>
          </ua:NodeId>
          <ua:NodeId>
            <ua:Identifier>ns=2;s="test02.sumon01"</ua:Identifier>
          </ua:NodeId>
        </NodeLookup>
      </ListOfPublishedNodes>
    </ua:XmlElement>
  </Extensions>

barnstee, 2017-02-01:
Argh, formatting got me, it should be:
`<Extensions>
    <ua:XmlElement>
      <ListOfPublishedNodes xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd">
        <NodeLookup>
          <ua:EndpointUrl>opc.tcp://localhost:27730</ua:EndpointUrl>
          <ua:NodeId>
            
            <ua:Identifier>i=2257</ua:Identifier>
          </ua:NodeId>
          <ua:NodeId>
            <ua:Identifier>ns=2;s="test02.sumon01"</ua:Identifier>
          </ua:NodeId>
        </NodeLookup>
      </ListOfPublishedNodes>
    </ua:XmlElement>
  </Extensions>
`

closed 2017-02-01 by barnstee

sumongithub, 2017-02-03:
I have modified the XML, but it is not appearing in the azure message.

**Modified Code**
`<Extensions>
    <ua:XmlElement>
      <ListOfPublishedNodes xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd">
        <NodeLookup>
          <ua:EndpointUrl>opc.tcp://localhost:27730</ua:EndpointUrl>
          <ua:NodeId>
            
            <ua:Identifier>i=2257</ua:Identifier>
          </ua:NodeId>
          <ua:NodeId>
            <ua:Identifier>ns=2;s="test02.sumon01"</ua:Identifier>
          </ua:NodeId>
        </NodeLookup>
      </ListOfPublishedNodes>
    </ua:XmlElement>
  </Extensions>`

Only values for i=2257 is apearing
**Transmitted message:**
`03-02-2017 08:35:19 PM> Device: [iotdevice2], Data:[{"HostName":"lenovopc","MonitoredItem":{"Id":"i=2257","Uri":"urn:LenovoPC:UnifiedAutomation:UaServerCpp"},"ClientHandle":2,"Value":{"Value":"2017-02-03T14:59:51.9473689Z","SourceTimestamp":"2017-02-03T14:59:51.9473689Z","ServerTimestamp":"2017-02-03T15:05:01.7765739Z"}}]Properties:
'source': 'mapping'
'content-type': 'application/opcua+uajson'`

But tag "test02.sumon01" is apearing in another OPC client

[image]