Azure/iot-edge-opc-publisher-v1 #13: Custom tags are not appearing in azure IoT device messages

sumongithub, 2017-02-06:
I have modified the XML, but it is not appearing in the azure message.

**Modified Code**
`<Extensions> <ua:XmlElement> <ListOfPublishedNodes xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd"> <NodeLookup> <ua:EndpointUrl>opc.tcp://localhost:27730</ua:EndpointUrl> <ua:NodeId>  <ua:Identifier>i=2257</ua:Identifier> </ua:NodeId> <ua:NodeId> <ua:Identifier>ns=2;s="test02.sumon01"</ua:Identifier> </ua:NodeId> </NodeLookup> </ListOfPublishedNodes> </ua:XmlElement> </Extensions>`

Only values for i=2257 is apearing
**Transmitted message:**
`03-02-2017 08:35:19 PM> Device: [iotdevice2], Data:[{"HostName":"lenovopc","MonitoredItem":{"Id":"i=2257","Uri":"urn:LenovoPC:UnifiedAutomation:UaServerCpp"},"ClientHandle":2,"Value":{"Value":"2017-02-03T14:59:51.9473689Z","SourceTimestamp":"2017-02-03T14:59:51.9473689Z","ServerTimestamp":"2017-02-03T15:05:01.7765739Z"}}]Properties: 'source': 'mapping' 'content-type': 'application/opcua+uajson'`

**But tag "test02.sumon01" is apearing in another OPC client**
[image]

barnstee, 2017-02-06:
Try without the double-quotes

sumongithub, 2017-02-06:
Tested, but problem remain same

barnstee, 2017-02-06:
Please query the node using the reference client from the opc foundation github:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/SampleApplications/Samples/Client.Net4
Then copy the node ID as displayed in the client.

sumongithub, 2017-02-07:
Tested with reference client from opc foundation github. Below is my observation-

It is considering only one tag which is configured first in the node of xml configuration file. e.g. if I configure 
`<ua:NodeId>
            <ua:Identifier>ns=2;s=test02.Sumon02</ua:Identifier>
            <ua:Identifier>ns=2;s=test02.Sumon01</ua:Identifier>
  </ua:NodeId>`

it will only consider "ns=2;s=test02.Sumon02", second one will not appear in the message.

I think it is a genuine defect. Please suggest

barnstee, 2017-02-08:
This is by design: Only 1 identifier is allowed per node tag. So the correct format would be:
` <NodeLookup>
          <ua:EndpointUrl>opc.tcp://TODO:Add uri here</ua:EndpointUrl>
          <ua:NodeId>
            <ua:Identifier>ua:Identifier>ns=2;s=test02.Sumon02</ua:Identifier>
          </ua:NodeId>
        </NodeLookup>
        <NodeLookup>
          <ua:EndpointUrl>opc.tcp://TODO:Add another uri here</ua:EndpointUrl>
          <ua:NodeId>
            <ua:Identifier>ua:Identifier>ns=2;s=test02.Sumon01</ua:Identifier>
          </ua:NodeId>
        </NodeLookup>`

closed 2017-02-08 by barnstee