Azure/iot-edge-opc-publisher-v1 #21: Building OPC UA client with .NET

maccence, 2017-03-08:
I Downloaded the code from https://github.com/Azure/iot-gateway-opc-ua and I found  in the lib folder a bloc note asking me to add these files to build the project: 
aziotsharedutil.lib
gateway.lib
nanomsg.lib
parson.lib

But I can't find them on the net.

I tried to do this by enable dotnet binding using these two commands:
build_dotnet.cmd
build.cmd --enable-dotnet-binding
But the result of the first command was : "wget is recognized as the name of a cmdlet, function or..."

See below some pictures illustrating my problems.

Can someone help me? Thank you.
[image]
[image]

606simps, 2017-03-22:
Looks like you will also need to clone a copy of IoT Gateway SDK and get the libs from it.
Refer to the last paragraph of [this](https://github.com/Azure/iot-gateway-opc-ua#running-the-module).

barnstee, 2017-03-22:
Yeah! This pain will go away soon as we're working on creating NuGet packages for all of this.

closed 2017-03-22 by barnstee