Azure/iot-edge-opc-publisher-v1 #24: Custom tags

janfl, 2017-04-12:
Is it possible to add custom tags to elements within the MonitoredItems section.
I like to add my own tags to be forwarded to the Iot Hub

barnstee, 2017-04-12:
Yes, just call encoder.WriteString().

closed 2017-04-12 by barnstee

janfl, 2017-04-21:
I need the value to be stored in the gateway configuration file preferable in the MonitoredItem section and if no custom action is performed the tag is just passed on to the IoTHub in the MonitoredItem_Notification method
Could be something like:
"MonitoredItems": [
              {
                "StartNodeId": "ns=2;s=Simulation Examples.Functions.Random2",
                "NodeClass": 2,
                "DisplayName": "Random2",
                "DiscardOldest": false,
                "CustomData": "key1:value1,key2:value2"
              }
            ]