Azure/iot-edge-opc-publisher-v1 #28: Error loading 'dotnetcore' thrown from GatewayInterop.CreateFromJson

brnkrygs, 2017-05-04:
Hi!

First, I'd like to say thank you for putting this tool/example together.  It has some exciting possibilities for us, and I especially appreciate the work you've done to get everything published up to NuGet.  The build experience getting this all going was *dramatically* improved in the recent release.

That said, we're hitting a problem running this project on some dedicated gateway hardware running Windows Server 2016 Standard.

Everything worked OK on my dev box with visual studio installed, but when I tried to run the project on our server, I got the following error right after the Gateway config file was successfully patched:

```
Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'dotnetcore': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at Microsoft.Azure.Devices.Gateway.NetCoreInterop.InitializeDelegatesOnNative(IntPtr createAddress, IntPtr receiveAddress, IntPtr destroyAddress, IntPtr startAddress)
   at Microsoft.Azure.Devices.Gateway.NetCoreInterop.InitializeDelegates()
   at Microsoft.Azure.Devices.Gateway.GatewayInterop.CreateFromJson(String file_path)
   at GatewayApp.NetCore.Program.Main(String[] args)
```

I tried running the dll published from our dev box, and I tried building right on the server with the `dotnet clr`.  I also tried configuring the Gateway project to emit an `.exe` instead of a `.dll`. No luck either way.  Any ideas?

I posted a gist with a log of the commands we ran here:
https://gist.github.com/brnkrygs/c76f9f1cede12f6f218a9e4860add342

A couple of potentially interesting things: 

 1. I'm using the latest stable `dotnet core` version of `1.0.3`
 2. In order to `dotnet restore` on the gateway box, I had to `dotnet migrate`, which reported success

Thanks again for your attention and effort!

barnstee, 2017-05-04:
It can't find the natvie libraries from the GW SDK NuGet package. You need to run dotnet publish on your server as specified in the "Building the Module" section of the readme, then dotnet run.

brnkrygs, 2017-05-04:
Hi @barnstee - thank you for the response!

Unfortunately, I did try publishing from the server... 

this is the `dotnet restore` command I ran:
https://gist.github.com/brnkrygs/c76f9f1cede12f6f218a9e4860add342#file-command-log-txt-L45

and this is the `dotnet publish` command I ran: 
https://gist.github.com/brnkrygs/c76f9f1cede12f6f218a9e4860add342#file-command-log-txt-L91

Does it matter that there was a `dotnet build` in-between there?

brnkrygs, 2017-05-05:
Just realized that I didn't really make this clear... the command log in the linked gist https://gist.github.com/brnkrygs/c76f9f1cede12f6f218a9e4860add342 is from on the gateway server.  All of those commands were run directly on the gateway itself.

brnkrygs, 2017-05-05:
Thanks for the suggestion @barnstee 

I downloaded the NetCoreSDK as close to 1.0.0 as I could get... this is what I found:

https://github.com/dotnet/core/blob/master/release-notes/download-archive.md

I downloaded and installed both the 32-bit and 64-bit SDK Installer from here:
.NET Core 1.0 SDK Preview 2 build 3121
https://github.com/dotnet/core/blob/master/release-notes/download-archives/1.0-preview2-download.md

and removed all traces I could find to my original sdk install.

Then, downloaded a fresh copy of the `iot-gateway-opc-ua` release from github.

`dotnet restore` worked out of the box this time - no `dotnet migrate` needed.

`dotnet publish` also worked, and looked like this:

```
PS C:\Support\iot-gateway-opc-ua-1.0.0\iot-gateway-opc-ua-1.0.0> dotnet publish .\src\GatewayApp.NetCore\
Publishing GatewayApp.NetCore for .NETCoreApp,Version=v1.0
Project IoTHubCredentialTools (.NETStandard,Version=v1.3) will be compiled because expected outputs are missing
Compiling IoTHubCredentialTools for .NETStandard,Version=v1.3

Compilation succeeded.
    0 Warning(s)
    0 Error(s)

Time elapsed 00:00:12.0512835
Project Opc.Ua.Publisher.Module (.NETStandard,Version=v1.3) will be compiled because dependencies changed
Compiling Opc.Ua.Publisher.Module for .NETStandard,Version=v1.3

Compilation succeeded.
    0 Warning(s)
    0 Error(s)

Time elapsed 00:00:12.2649834
Project GatewayApp.NetCore (.NETCoreApp,Version=v1.0) will be compiled because dependencies changed
Compiling GatewayApp.NetCore for .NETCoreApp,Version=v1.0

Compilation succeeded.
    0 Warning(s)
    0 Error(s)

Time elapsed 00:00:12.0037659

publish: Published to C:\Support\iot-gateway-opc-ua-1.0.0\iot-gateway-opc-ua-1.0.0\src\GatewayApp.NetCore\bin\Debug\netcoreapp1.0\publish
Published 1/1 projects successfully
```

I jumped into the publish directory:

`PS C:\Support\iot-gateway-opc-ua-1.0.0\iot-gateway-opc-ua-1.0.0> cd .\src\GatewayApp.NetCore\bin\Debug\netcoreapp1.0\publish\`

and ran the published dll, but got the same error:

```
PS C:\Support\iot-gateway-opc-ua-1.0.0\iot-gateway-opc-ua-1.0.0\src\GatewayApp.NetCore\bin\Debug\netcoreapp1.0\publish> dotnet .\GatewayApp.NetCore.dll IotHub "<our connection string>"
Target system is Windows.
Target system is 64-bit.
Gateway config file patched with application name: IotHub
Attemping to register ourselves with IoT Hub using owner connection string: <our connection string>
Attemping to read connection string from secure store with certificate name: IotHub
Attemping to configure publisher with connection string: <our connection string>
Gateway config file patched with IoT Hub name: <our iot hub name>

Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'dotnetcore': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at Microsoft.Azure.Devices.Gateway.NetCoreInterop.InitializeDelegatesOnNative(IntPtr createAddress, IntPtr receiveAddress, IntPtr destroyAddress, IntPtr startAddress)
   at Microsoft.Azure.Devices.Gateway.NetCoreInterop.InitializeDelegates()
   at Microsoft.Azure.Devices.Gateway.GatewayInterop.CreateFromJson(String file_path)
   at GatewayApp.NetCore.Program.Main(String[] args)
PS C:\Support\iot-gateway-opc-ua-1.0.0\iot-gateway-opc-ua-1.0.0\src\GatewayApp.NetCore\bin\Debug\netcoreapp1.0\publish>
```
Did I grab the right version of .NetCore1.0.0?  Any other thoughts?  We also have a full version of the .NET Framework installed on that box... somewhere around 4.6 (supporting netstandard1.3), if that matters.

barnstee, 2017-05-08:
Hmm, wired. I actually recommend running Docker on the GW machines, follow the instructions here:
https://docs.microsoft.com/en-us/azure/iot-suite/iot-suite-connected-factory-gateway-deployment
But of course this is no good to you if you have made code changes.
The only other thing is for you to check if, after running the managed app once, you see the GW's runtime DLLs in the publish directory, i.e. there should be a dotnetcore.dll and other stuff like IoTHub.dll etc. in there...

brnkrygs, 2017-05-08:
Thanks for the link, I'll take a crack at it!  There's no code changes (yet), so I'm hopeful :)

barnstee, 2017-05-10:
did that work?

brnkrygs, 2017-05-10:
I was able to load Docker onto the Gateway, and run the test container.  Haven't had a chance to revisit it, hoping to pick it back up in about 2 hours.  Will keep you posted!

brnkrygs, 2017-05-10:
Quick update for you... I had to install Docker for Windows Server instead of Docker for Windows (since the underlying OS is Windows Server 2016, Docker for Windows wouldn't install) and the documentation is... harder to find.  I'm getting an error trying to mount the local "Docker" folder I created when following step 2 in the docs in the link provided above.  

`Error response from daemon: Invalid bind mount spec`

Doing some research on how to resolve that so I can get the first container running.

barnstee, 2017-05-10:
did you share the drive with docker first?

brnkrygs, 2017-05-10:
Trying to figure out how to do that... "Docker for Windows Server" didn't install the handy little "settings" app (at least that I can find) that gets installed for "Docker for Windows".

barnstee, 2017-05-10:
Ah. :-)

barnstee, 2017-05-18:
I think I now know what the issue with the upgraded soluion was: The .Net team changed the way native libraries in NuGet packages are handled and this is why they are no longer picked up and copied to the right place during publishing. So the NuGet package I currently use only works with .NetCore 1.0.0.

brnkrygs, 2017-05-19:
Thanks for the update! I'll take a shot with .NetCore 1.0.0 specifically.

I'm assuming that the `1.0 with SDK Preview 2 build 3121` download on the download archives page is the right one...
https://github.com/dotnet/core/blob/master/release-notes/download-archive.md

I've still been fighting with mounting volumes on Windows Server 2016's version of Docker, so this comes as pretty good news actually :)

barnstee, 2017-05-19:
Yeah...

barnstee, 2017-05-24:
Any luck?

barnstee, 2017-05-29:
Closing. Please reopen if still an issue.

brnkrygs, 2017-05-30:
I'll do that.  I apologize for the radio silence, thank you for your attention so far.

brnkrygs, 2017-05-31:
Was able to try the oldest LTS release from the dotnet core archives page.  Same error as before :(

Here's output from the various `dotnet` invocations, showing build versions.  I'm not sure its really .NetCore 1.0.0 - looks like 1.0.1 to me.  I'm scouring GitHub to see if I can find a 1.0.0 release anywhere.  If you have a link, I'd really appreciate it!

```
PS C:\Users\Administrator> dotnet

Microsoft .NET Core Shared Framework Host

  Version  : 1.0.1
  Build    : cee57bf6c981237d80aa1631cfe83cb9ba329f12

PS C:\Users\Administrator> dotnet --version
1.0.0-preview2-003121

PS C:\Users\Administrator> Get-Command dotnet

CommandType     Name                                               Version    Source
-----------     ----                                               -------    ------
Application     dotnet.exe                                         1.0.1.4500 C:\Program Files\dotnet\dotnet.exe
```

barnstee, 2017-06-01:
https://github.com/dotnet/core/blob/master/release-notes/download-archive.md

brnkrygs, 2017-06-01:
Yeah, that's what I used unfortunately.

barnstee, 2017-06-01:
OK, I'm going to upgrade to VS2017/.NetCore1.1.2 ASAP, you're not the only one who has this issue, so time to pull the pin...

brnkrygs, 2017-06-01:
Since that's the case, I'm seeing the same error with .Net Core 1.0.0... should we re-open this issue?

Something that might help is if the system logged out the paths it was searching when looking for "DLL 'dotnetcore'", kind of like how ASP.NET MVC logs the paths it searched when there's a missing `SomeView.cshtml` file.  Then I could at least manually drop the DLLs in the right spot.

Is something like that doable within the scope of this app?

barnstee, 2017-06-01:
No as it is done in the GW runtime and not in the app...

brnkrygs, 2017-06-01:
Kinda figured it would be pretty buried :)

I'll watch for a release against VS2017/.NetCore1.1.2!

Again, appreciate your continued attention.

barnstee, 2017-06-08:
OK, release is done, please try it. Docker images still building...

barnstee, 2017-06-08:
Docker image (V1.1.1) now also available on Dockerhub.

brnkrygs, 2017-06-09:
Will take a shot!  Thank you!

brnkrygs, 2017-06-09:
UPDATE WORKED!

I downloaded/installed the (as of today) latest stable [NetCore release (1.1)](https://www.microsoft.com/net/core), ran the restore and publish commands and got past loading my `gatewayconfig.json`!

`c:\project_root>dotnet restore`
`c:\project_root>dotnet publish .\src\GatewayApp.NetCore`
`c:\project_root>dotnet .\src\GatewayApp.NetCore\bin\Debug\netcore1.1\publish\GatewayApp.NetCore.dll`

all worked.

Looks like I'm in great shape now, thank you!!

closed 2017-06-09 by brnkrygs