Azure/iot-edge-opc-publisher-v1 #29: Error: There is no PublishedNodes.json

PedroFloresGuillen, 2017-05-10:
I followed the building module, and when i was going to configure it, I found there was no publishedNodes.json neither in the proyect nor the folder.

When I was restoring, those warmings appeared

[image]

I published it without problems.

My dotnet --version
1.0.0-preview2-003131

So, maybe the fault is in the publish part?

PedroFloresGuillen, 2017-05-10:
Nevermind, I created the file and it solved the problem.

barnstee, 2017-05-10:
No error. The publishednodes.json file is *automatically generated and persisted to disk* as soon as you call the OPC UA server interface built-in to the publisher (publishnode method). You can of course create one yourself manually, if you want.

rish11, 2017-06-05:
@PedroFloresGuillen where did you created that file ?

PedroFloresGuillen, 2017-06-05:
To manually create the file, I used Visual Studio to create a Json file named publishednodes in GatewayApp.NetCore

rish11, 2017-06-05:
@PedroFloresGuillen  I am new to VS can you guide me step by step on how to configure the module ?

PedroFloresGuillen, 2017-06-05:
Of course, in order to have a publishednodes.json you should do these steps.
[image]

[image]

[image]

rish11, 2017-06-05:
@PedroFloresGuillen I have already done this. I'm asking steps for Configuring the gateway

PedroFloresGuillen, 2017-06-05:
You should explain more what do you want. But, if you mean passing the args, you have 2 args to pass.
Aplication Name: This can be wathever you want, it will be the name of the gateway.
Iot Hub address: This is the address of the hub you want to link the gateway.

[image]

rish11, 2017-06-05:
@PedroFloresGuillen  ohk 
so what should be the IoT hub address? Should I take it from azure portal ? Is it the connecting String ?

PedroFloresGuillen, 2017-06-05:
Yes, the conection String from azure of the IoT hub you want to use.

rish11, 2017-06-05:
I have passed the arguments. And now when I'm running it, It's showing dll file missing error
[image]

PedroFloresGuillen, 2017-06-05:
[image]

Did you build edge this way? Also, what version do you have of the publisher and gateway?

rish11, 2017-06-05:
No I didn't build it this way,I have just press F5 to run 
I have the latest version of both

PedroFloresGuillen, 2017-06-05:
When you build the gateway, you should build it with --enable-dotnet-binding, or the module wont work.

rish11, 2017-06-05:
but there isn't any tool folder in iot-edge-opc-publisher
C:\Users\rishabhgupta_pwc\Desktop\OPCCLIENT> 
This is the directory where all the files are. what i have to do next ?

PedroFloresGuillen, 2017-06-05:
iot-edge-opc-publisher is a module for Iot-edge. you have to clone iot edge, build it with the options, and then you will be able to use the module.

rish11, 2017-06-05:
I have done as yous aid while building I got these two errors. 
[image]
[image]

PedroFloresGuillen, 2017-06-05:
Delete iot edge master folder
do: git clone https://github.com/Azure/iot-edge.git
do: cd: iot-edge/tools
do: build.cmd --enable-dotnet-binding

rish11, 2017-06-05:
same error is showing

PedroFloresGuillen, 2017-06-05:
well, i dont know why it happens then, never happened to me, you should open a new issue and try your luck there.

rish11, 2017-06-06:
@PedroFloresGuillen  
Hey, My problem has been solved. 
I the project is running and I have connected it to IoT Hub. MY OPC Server is active but I can't see the data. Where should I see it ?

PedroFloresGuillen, 2017-06-06:
That is what im working right on, if I manage to make something viable I will tell you.

barnstee, 2017-06-06:
IMPORTANT: You don't need to clone IoT-Edge. We have NuGet packages for IoT-Edge now.

rish11, 2017-06-07:
yup.
My iot-edge-opc-publisher is working. 
But now the problem how should I read the data from OPC client ? 
Also from where you have taken the Mock OPC server please tell me it will be really helpfull.

PedroFloresGuillen, 2017-06-07:
https://github.com/OPCFoundation/UA-.NETStandardLibrary (here is where i got the mock server)

To see data from a server, you can use a client that connects to it, like UAExpert

rish11, 2017-06-07:
Then how are you visualizing the data in IoT hub ? 
And I have endpoint url of an active server and I have put it in PublishedNodes.json 
But I dont know how to see the tags in IoT Hub..

PedroFloresGuillen, 2017-06-07:
Oh, currently im not visualizing the data in IoT hub, its what im looking too.

barnstee, 2017-06-07:
https://github.com/Azure/azure-iot-sdk-csharp/tree/master/tools/DeviceExplorer

barnstee, 2017-06-07:
Simple website for telemetry: https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/OpcUaWebTelemetry

barnstee, 2017-06-07:
Full connected factory preconfigured solution using Publisher: www.azureiotsuite.com

closed 2017-06-08 by barnstee

f17jwr, 2017-07-18:
@rish11 I am still having issues even after building iot-edge, as you did - how did you overcome this?
it just says my runtime folders are not found