Azure/iot-edge-opc-publisher-v1 #30: Trying to build module in Docker container - Missing project.json

faister, 2017-05-11:
I followed the steps to build the module in a Docker container using the Dockerfile provided. From the root of the repo, in a console, I type:
docker build -t gw .

`C:\Demos\iot-edge-opc-publisher>docker build -t gw .
Sending build context to Docker daemon 42.86 MB
Step 1/9 : FROM microsoft/dotnet:1.0-sdk-projectjson
 ---> 5725a9993230
Step 2/9 : COPY / /build
 ---> Using cache
 ---> 28982d329215
Step 3/9 : WORKDIR /build
 ---> Using cache
 ---> 0e2194a2d728
Step 4/9 : RUN dotnet restore
 ---> Using cache
 ---> 5f8617b9c66a
Step 5/9 : WORKDIR /build/src/GatewayApp.NetCore
 ---> Using cache
 ---> 568b80adec09
Step 6/9 : RUN dotnet publish
 ---> Running in a9e1e7b4fe30
'/build/src/GatewayApp.NetCore' does not contain a project.json file
The command '/bin/sh -c dotnet publish' returned a non-zero code: 1`

faister, 2017-05-11:
Resolved the error. But it still needs to be fixed in this repo. I had to copy the project.json file from the backup\src folders to their respective folders in \src. Successfully built the container this way.

barnstee, 2017-05-11:
The project.json file is there: https://github.com/Azure/iot-edge-opc-publisher/blob/master/src/GatewayApp.NetCore/project.json
If it was removed, you probably upgraded the project with VS2017, correct? VS2017 unfortunately removes the project.json file during upgrade of the project.

closed 2017-05-18 by barnstee