Azure/iot-edge-opc-publisher-v1 #32: Functions of IoTHub module and OPC UA Pub-Sub specs realization.

JayeshThamke, 2017-05-22:
Hello, 
I have two questions, 
1. What are the functions of IoTHub module in the Gateway module apart from publishing UA Nodes data to IoTHub in Azure cloud? Is there any documentation?
2. There is a RC OPC UA Pub-Sub released specs, is the Azure OPC UA Publisher module complying to that spec because there are several nodes (type nodes) specific to pub-sub specifications those are not available in SDK. Please correct me if I am wrong. Thanks,

barnstee, 2017-05-24:
1. All it does is publish modes data and yes, it translates uabinary to json.
2. The publisher is compliant when it comes to the json format (it uses the json encoder from the OPC Foundation), but it does not provide an implementation of the pub-sub node spec, as this is not needed for sending data to the cloud and I came up with a much simpler node set of just 3 nodes. :-)

JayeshThamke, 2017-05-25:
@barnstee thanks for reply, 
For my thesis project I am developing a OPC Publisher which can of course publish data of my Ua Server to Azure IoTHub (the normal case that you have designed) and also the same OPC Publisher component I want to use in a Pub-Sub model with other OPC Subscribers (UA Clients) with minimum changes. For my second case, will I need an OPC Publisher module to publish data to Broker in Ua JSON format or in Ua Binary format? I referred to UA PubSub Specs?

barnstee, 2017-05-29:
For Pub-Sub, you can use either UA Binary or JSON.

JayeshThamke, 2017-05-29:
Ok, Thank you very much for reply,
I am not whether I (issue author) am suppose to close this issue thread or you. For me, I got the required inputs till now. Thanks 
:+1:

closed 2017-05-29 by barnstee