Azure/iot-edge-opc-publisher-v1 #33: Running opc publisher module as part of iot edge

jild, 2017-05-23:
Hi,

i've been struggling to get a working config to get the opc module to work in azure iot edge. Is this possible? I tried to use the donet_binding_sample and the dotnet_core_module_sample and fed it with my own configuration (see below). Once with the coreclrpath once without (referring to: https://github.com/Azure/iot-edge/issues/273). But I always get the error:

> Error: Time:Tue May 23 14:10:39 2017 File:C:\Users\user\Downloads\edge-scratch\iot-edge5\core\adapters\dynamic_library_windows.c Func:_DynamicLibrary_LoadLibrary Line:17 Error Loading Library. Error code is:  126                           Error: Time:Tue May 23 14:10:39 2017 File:C:\Users\user\Downloads\edge-scratch\iot-edge5\bindings\dotnetcore\src\dotnetcore.cpp Func:?DotNetCore_Create@@YAPAXPAUBROKER_HANDLE_DATA_TAG@@PBX@Z Line:116 Failed to Load .NET Core CLR Library.  Error: Time:Tue May 23 14:10:39 2017 File:C:\Users\user\Downloads\edge-scratch\iot-edge5\core\src\gateway_internal.c Func:_gateway_addmodule_internal Line:446 Module_Create failed.                                                           Error: Time:Tue May 23 14:10:39 2017 File:C:\Users\user\Downloads\edge-scratch\iot-edge5\core\src\gateway_createfromjson.c Func:_Gateway_CreateFromJson Line:78 Failed to create gateway using lower level library.                            failed to create the gateway from JSON             

      

Opc.Ua.Publisher.Module.dll iothub.dll gateway.dll dotnetcore.dll etc. all lying in the directory of the executable. Im running this from inside visual studio and from visual studio developer command prompt (like <sample> <config.json>)
```
{
  "modules": [
    {
      "name": "OpcUa",
      "loader": {
        "name": "dotnetcore",
        "type": "dotnetcore",
        "configuration": {
          "binding.coreclrpath": "C:\\Program Files\\dotnet\\shared\\Microsoft.NETCore.App\\1.1.2\\coreclr.dll",
          "binding.trustedplatformassemblieslocation": "C:\\Program Files\\dotnet\\shared\\Microsoft.NETCore.App\\1.1.2"
        },
        "entrypoint": {
          "assembly.name": "Opc.Ua.Publisher.Module",
          "entry.type": "Opc.Ua.Publisher.Module"
        }
      },
      "args": {
        "Configuration": {
          "ApplicationName": "<ReplaceWithYourApplicationName>",
          "ApplicationType": "ClientAndServer",
          "ApplicationUri": "urn:localhost:microsoft:publisher"
        }
      }
    },
    {
      "name": "IoTHub",
      "loader": {
        "name": "native",
        "entrypoint": {
          "module.path": "iothub.dll"
        }
      },
      "args": {
        "IoTHubName": "<ReplaceWithYourIoTHubName>",
        "IoTHubSuffix": "azure-devices.net",
        "Transport": "AMQP"
      }
    }
  ],
  "links": [
    {
      "source": "OpcUa",
      "sink": "IoTHub"
    }
  ]
}

```

Many thanks

barnstee, 2017-05-24:
Yes, the Publisher supprots IoT Edge. The problem is that right now, it's built using .Net Core 1.0.0 and you've got .Net Core 1.1.2 installed. If you install the old .Net Core 1.0.0 SDK, all will be well. Also, I'm in the provess of upgrading this to .Net 1.1.0, which is the last SDK that supports project.json files. Eventually I will upgrade to .csproj files (VS2017) and .Net 1.1.2.

barnstee, 2017-05-24:
Also, you can simply use the managed GW app, then you don't need to specify a .Net Core CLR location at all as this will be loaded by the managed GW app automatically: https://github.com/Azure/iot-edge-opc-publisher/tree/master/src/GatewayApp.NetCore

jild, 2017-05-28:
Hi thank you very much for your answer. The approach directly in IoT edge doesn't seem too work for me, same error even with 1.0.4 (does it really have to be 1.0.0 ?). Im still trying to track down the problem.

The managed GW app works, thank you. In order to assign an azure device to the module, I probably need to configure the Identity map module in gatewayconfig.json?

barnstee, 2017-05-29:
Yep, 1.0.0 only for now. For registering a device, this is done automatically if you pass in an iothubowner connection string as second argument.

barnstee, 2017-05-30:
Closing this issue. Please reopen if your questions haven't been answered yet.

closed 2017-05-30 by barnstee