Azure/iot-edge-opc-publisher-v1 #34: "PublisherModel.csv" file is not generated in Opc.Ua.Publisher.Module folder to generate Model

JayeshThamke, 2017-05-24:
Hi,
I want to generate a OPC UA Publisher NodeSet. I am using executing BuildServerModel.cmd from PowerShell, it always throws error 

`Opc.Ua.ModelCompiler.exe not found
cannot compile the station models`

I have already copied Opc.Ua.ModelCompiler.exe in the current directory, i.e. C:\github_repos\iot-gateway-opc-ua\src\Opc.Ua.Publisher.Module

Then I tried with executing Opc.Ua.ModelCompiler.exe through PowerShell, then came to know there is no PublisherModel.csv file in .\Opc.Ua.Publisher.Module folder.

I am not sure if I did some wrong.

Thanks,
Jayesh

barnstee, 2017-05-24:
The CSV is generated by the ModelCompiler, if you specify the correct argument (-cg). The ModelCompiler needs to be on the path. Why are you making a new uanodes file anyway?

JayeshThamke, 2017-05-25:
Thanks for reply @barnstee ,

> Why are you making a new uanodes file anyway?
Well, in my university thesis project I want to add some more features to OPC Publisher. I am considering it by sub-typing PublisherType and/or the custom methods like PublishNodeMethodTypes and UnPublishNodeMethodTypes etc. I hope for this extension I would need to modify PublisherModel.xml and create a new uanodes file, please correct me if I am wrong.  

Also, is the namespace used in PublishModel.xml i.e. 
`http://opcfoundation.org/UA/Publisher/` 
is it a standard namespace for PubSub NodeSet? 
Thanks again
Jayesh

barnstee, 2017-05-29:
Yes, in that case, you need to generate a new nodes file.
The namespace is just something I cmae up with for the Publisher, it is not a standard namespace.

JayeshThamke, 2017-05-29:
Thanks for reply, I have got the answers.

closed 2017-05-29 by barnstee