Azure/iot-edge-opc-publisher-v1 #38: Opc Publisher endpoint wont be find by its name

PedroFloresGuillen, 2017-06-06:
I have IoT Edge up and running with the module of OPC, but when I try to get to the endpoint of the server using a client (opc.tcp://<name>:62222/UA/Publisher) it wont find it. But when I connect to opc.tcp://localhost:62222/UA/Publisher it will appear without problems, being able to connect to it.

[image]

barnstee, 2017-06-06:
opc.tcp://:62222/UA/Publisher is not a valid URL. This issue appear to be related to UA Expert, please check with Unified Automation if they are able to parse this type of URL.

PedroFloresGuillen, 2017-06-07:
Sorry, i meant opc.tcp://Gateway:62222/UA/Publisher

barnstee, 2017-06-07:
Do you have a hosts file entry for "Gateway" on your system. What happens when you call "ping Gateway" from command prompt?

PedroFloresGuillen, 2017-06-08:
Oh, haven't thought of that, added it and now it works appropiately, sincerely thanks you.

barnstee, 2017-06-08:
not a problem, no.

closed 2017-06-08 by PedroFloresGuillen