Azure/iot-edge-opc-publisher-v1 #42: publishednodes.json

khilscher, 2017-06-16:
I am trying to create/populate the publishednodes.json file. Exactly how is this file is auto-generated? It sounds from the Readme I need to use an OPC UA client to connect and do something?

If I need to connect to the built-in OPC Server, what's the endpoint URL. On the console I see...

```Starting server on endpoint opc.tcp://khgateway:62222/UA/Publisher...```

Is this the complete endpoint URL? The ellipses (...) seem to indicate there's more to the URL.

Also, can we simply create this file manually? If so, where does the file need to reside?

khilscher, 2017-06-16:
I looked at Issue #29 and added publishednodes.json to the GatewayApp.NetCore project, rebuilt the docker image using ```docker build -t gw .``` and ran it using ```docker run -it --rm gw <applicationName>```. The console shows the following error:

```Opc.Ua.Publisher.Module: Nodes file loading failed with: Could not find file '/build/src/GatewayApp.NetCore/bin/Debug/netcoreapp1.1/publishednodes.json'.```

Also running it using the docker option ```-e \_GW\_PNFP="/shared/publishednodes.JSON"``` with the file in the shared directory on the docker host  also results in the same error.

Was able to connect to the OPC Server using the UaExpert OPC client but am unsure how to configure it this way.

PedroFloresGuillen, 2017-06-19:
You can create the file manually. Just add the file to "C:\Users \(user)\Desktop\Edge\iot-edge-opc-publisher\src\GatewayApp.NetCore\bin\Debug\netcoreapp1.1" and manually file the information.

khilscher, 2017-06-19:
That worked @PedroFloresGuillen . Not sure why I didn't think of it.

closed 2017-06-19 by khilscher