Azure/iot-edge-opc-publisher-v1 #43: Build Issues with VS 2017

khilscher, 2017-06-19:
1. ```git clone https://github.com/Azure/iot-edge-opc-publisher.git```
1. Open Opc.Ua.Publisher.Module.sln in VS 2017
1. Under Properties of GatewayApp.Netcore -> Debug I added App Arguments: <AppName> <IoTHubConnString>
1. GatewayApp.Netcore -> Publish -> 
 1. Target Location: \bin\Release\PublishedOutput
 1. Configuration: Release
 1. Target Framework: netcoreapp1.1
 1. Target Runtime: win10-x64
1. After publish I hit F5 and it complains it cannot find gatewayconfig.json. Same as Issue #40 
1. Copied gatewayconfig.json to src\GatewayApp.NetCore\bin\Debug\netcoreapp1.1 and updated placeholder values. 
    - **Question - why is this required when I am passing in <AppName> <IoTHubConnString> to the GatewayApp.NetCore module?**
1. Hitting F5 the compile when further but now am encounting the following error:

```
Opc.Ua.Publisher.Module: Attemping to connect to servers...
Opc.Ua.Publisher.Module: Created.
Error: Time:Mon Jun 19 14:34:12 2017 File:C:\agent\_work\7\s\iot-edge\core\adapters\dynamic_library_windows.c Func:DynamicLibrary_LoadLibrary Line:17 Error Loading Library. Error code is:  126
Error: Time:Mon Jun 19 14:34:12 2017 File:C:\agent\_work\7\s\iot-edge\core\src\module_loaders\dynamic_loader.c Func:DynamicModuleLoader_Load Line:72 DynamicLibrary_LoadLibrary() returned NULL for module iothub.dll
Error: Time:Mon Jun 19 14:34:12 2017 File:C:\agent\_work\7\s\iot-edge\core\src\gateway_internal.c Func:gateway_addmodule_internal Line:400 Failed to add module because the module could not be loaded.
Opc.Ua.Publisher.Module: All sessions closed.
Error: Time:Mon Jun 19 14:34:12 2017 File:C:\agent\_work\7\s\iot-edge\core\src\gateway_createfromjson.c Func:Gateway_CreateFromJson Line:78 Failed to create gateway using lower level library.
.NET Core Gateway failed to initialize. Please make sure you have published the GatewayApp.NetCore app to make sure the depend DLLs are available!
```

PedroFloresGuillen, 2017-06-20:
Hello Khilscher
not being able to find gatewayconfig.json after publishing may be because you didnt publish it in is respective folder.
If you are using the Debug, you should publish in "iot-edge-opc-publisher\src\GatewayApp.NetCore\bin\Debug\netcoreapp1.1" instead of the default while leaving the original file in the "GatewayApp.NetCore" folder and in the project.

khilscher, 2017-06-20:
I was able to work around the issue by:

- cd to ```src\GatewayApp.NetCore\bin\Release\PublishOutput```
- ```dotnet GatewayApp.Netcore.dll <appname> <IotHub Conn String>```

I think the documentation around this project really needs to be more comprehensive. It's been super painful to get this project code running. I was able to get a minimum OPC UA to Azure IoT Hub GW running in a couple hours (https://github.com/khilscher/OPCUA-Azure-IoTHub-GW) using the OPC Foundation client code. By improving the documentation on this code it would greatly improve its usability for Proof of Concepts.

closed 2017-06-20 by khilscher