Azure/iot-edge-opc-publisher-v1 #46: Running Windows Docker container - Unable to load DLL 'dotnetcore'

faister, 2017-06-21:
I built a Windows Docker container, but when I run it, it is unable to load the .NET Core DLL. I explicitly set the PATH to the Windows runtime for .NET Core DLL. 
ENV PATH=/build/src/GatewayApp.NetCore/bin/Debug/netcoreapp1.1/runtimes/win-x64/native;/build/src/GatewayApp.NetCore/bin/Debug/netcoreapp1.1/

Target system is Windows.
Microsoft Windows 10.0.14393
Target system is 64-bit.
Gateway config file patched with application name: publisher.opc.corp.contoso
Attemping to register ourselves with IoT Hub using owner connection string: HostName=aufactorydb141.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=QLwsV6I3eZ1i0kssm73o4qsy3NCLRxY/kc0oZyVGzn4=
Attemping to read connection string from secure store with certificate name: publisher.opc.corp.contoso
Attemping to configure publisher with connection string: HostName=aufactorydb141.azure-devices.net;DeviceId=publisher.opc.corp.contoso;SharedAccessKey=zsLcrwmpmqVoMfSn55DbVuXvBm8uPLqrSl5mwVLc+mE=
Gateway config file patched with IoT Hub name: aufactorydb141

Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'dotnetcore': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at Microsoft.Azure.Devices.Gateway.NetCoreInterop.InitializeDelegatesOnNative(IntPtr createAddress, IntPtr receiveAddress, IntPtr destroyAddress, IntPtr startAddress)
   at Microsoft.Azure.Devices.Gateway.NetCoreInterop.InitializeDelegates() in C:\agent\_work\7\s\iot-edge\bindings\dotnetcore\dotnet-core-binding\Microsoft.Azure.Devices.Gateway\NetCoreInterop.cs:line 279
   at GatewayApp.NetCore.Program.Main(String[] args) in C:\build\src\GatewayApp.NetCore\Program.cs:line 106

barnstee, 2017-06-21:
I don't think this ill work. It is clearly finding the runtime (otherwise it wouldn't run the app at all) but the IoT Edge can't find it. You specify the path to the runtime via the gatewayconfig.json file. See here for an example:
https://github.com/Azure/iot-edge/blob/master/samples/dotnet_core_module_sample/src/dotnet_core_module_sample_win.json

faister, 2017-06-22:
Thanks for the pointer. I specified the path to the dotnetcore.dll according to the example above, tried it out under dotnet run, it works.
But I still ran into the same issue of not being able to load DLL 'dotnetcore'. As you can see in the screen capture [image], I have specified the path to coreclr.dll and I have also listed the specified directory to make sure that coreclr.dll is in it.

barnstee, 2017-07-07:
Fixed in recent checkin.

closed 2017-07-07 by barnstee