Azure/iot-edge-opc-publisher-v1 #53: Could not send an open secure channel request

carlwouters, 2017-07-11:
Specifications:
- Windows PC
- Kepware OPC UA sample server
- Certificates should be ok (after issues with SHA1 certificates)

When trying to get the OPC UA publisher working, we are getting an error:

12:13:55.647 ***EXCEPTION*** BadTcpInternalError 'Could not send an Open Secure Channel request.'
12:13:55.651 ***EXCEPTION*** BadTcpInternalError 'Error establishing a connection: Could not send an Open Secure Channel request.'

I’ve added the log file:

[DigitalTwinTest.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1138089/DigitalTwinTest.log.txt)

PedroFloresGuillen, 2017-07-11:
Have you tried using the .net sample server?

carlwouters, 2017-07-12:
When trying to connect to the sample server with UAExpert, I get this error:

[image]

The UAExpert certificate is in the UA Application certs folder. (not in the rejected anymore)

carlwouters, 2017-07-12:
Seems that the certificate made by the iot edge publisher was a protected one with a key. Removing that solved the issue.

closed 2017-07-12 by carlwouters