Azure/iot-edge-opc-publisher-v1 #56: publishednode.json doesn't contain namespace URI

kiranpradeep, 2017-07-17:
As per configuring the application section( [link](https://github.com/Azure/iot-edge-opc-publisher#configuring-the-application) ), the `publishednodes.json` file is expected to contain nodeid and not namespace URI. The namespace index is just valid for a connection to the server. It may change for the next connect( [link](http://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l2_ua_node_ids.html) ). Shouldn't the publishdnode.json file format be updated to contain namespace URI ?

Or does in real-world, the namespace index will remain constant ?

barnstee, 2017-07-18:
A few things:
1. We're not using the Unified Automation SDK, so Unified Automation docs may not apply to the Publisher. it does here, fortunately. The OPC UA .Net Standard stack (which we do in fact use) documentation can be found here: http://opcfoundation.github.io/UA-.NETStandardLibrary
2. Yes, the namespace index is a constant and doesn't change between connections.
3. The namespace index and the node ID uniquely identify a UA node within an UA server, so the format of the publishednodes.json is correct.

kiranpradeep, 2017-07-20:
@barnstee. Thanks for the reply. But my doubts remain. My understanding of OPC-UA could be wrong. Requesting clarification on doubts. To make sure we are on the same page, I am speaking about the OPC-UA client aspect of this Azure module. Also, my understanding was that unified automation docs were the final word on OPC-UA - Is that wrong?

1. It doesn't matter what SDK this module uses. Since this module has to connect with OPC-UA servers( which complies with OPC-UA ), shouldn't it comply with Unified Automation docs?
2. No. As per unified automation docs, namespace index is not constant. ([link](http://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l2_ua_node_ids.html)) 
3. Again No. As per this [link](https://forum.unified-automation.com/topic1697.html),node-id and index can uniquely identify a node only **per** session. The same client connecting **again** to the same server could receive a different index for namespaces.

barnstee, 2017-08-01:
The Unified Automation docs are good, but I wouldn't consider them the "final world", no. They represent the docs of the implementation by Unified Automation, nothing more. The OPC UA specs from the OPC Foundation are the "final word".
For 1.: I agree that when the Unified Automation docs talk about generic behavior of an OPC UA server, they should generally apply to all OPC UA servers, but I would read the UA specs instead.
For 2. & 3.: Yes, a namespace index can change if the server is reconfigured (new nodes added, old ones removed). In practice, this doesn't happen (once a server is configured it stays that way) and therefore the namespace index doesn't change. I agree with you that a more generic solution would not hard-code the namespace index in the publishednodes.json. In practice this is sufficient. Do you want to take a stab at changing the implementation and sending out a pull request?

kiranpradeep, 2017-08-02:
@Thanks. Sorry, I am not familiar with C# and so will find it hard to submit a pull request. Should I leave this ticket open?

barnstee, 2017-08-02:
I'd leave it open, yes.

barnstee, 2017-08-02:
We'll see if we can squeeze it in...

hansgschossmann, 2017-09-02:
@kiranpradeep thanks for pointing this out in detail. i will change the implementation to support a per session namespace index lookup based on the namespace URI for the node, which needs to be specified in the publishednodes.json.

hansgschossmann, 2017-11-14:
@kiranpradeep: fixed a while back in 2.1.0 already.

closed 2017-11-14 by hansgschossmann