Azure/iot-edge-opc-publisher-v1 #57: cannot find Runtimes folder

f17jwr, 2017-07-18:
[image]

I am having issues resolving this problem, any help would be greatly appreciated.

f17jwr, 2017-07-18:
Upon further digging I have realised that a 'runtimes' folder is missing from my GatewayApp.NetCore folder and I am unsure why.

barnstee, 2017-07-18:
Can you use the latest code from master? GatewayApp.NetCore is now no longer needed.

closed 2017-07-18 by barnstee

f17jwr, 2017-07-18:
What code do you mean?

barnstee, 2017-07-18:
The publisher code.

f17jwr, 2017-07-19:
That still isn't helpful.

barnstee, 2017-07-19:
Please sync to the latest Publisher code in the master branch, i.e. git pull from master branch. You will see that  GatewayApp.NetCore has been removed as it is no longer needed, along with the code to find the runtimes folder.

f17jwr, 2017-07-19:
Ah ok thankyou.