Azure/iot-edge-opc-publisher-v1 #58: Trouble Adding a New Gateway to a Connected Factory

KZeronimo, 2017-07-18:
We are attempting to add our own gateway to a connected factory instance following this [guidance](https://docs.microsoft.com/en-us/azure/iot-suite/iot-suite-connected-factory-gateway-deployment)

We have accomplished all of the sanity checks and are failing at step 5

> When you have successfully completed these local tests, browse to the Connect your own OPC UA Server page in the connected factory solution portal. Enter the publisher endpoint URL (opc.tcp://publisher.<your fully qualified domain name>:62222) and click Connect. You get a certificate warning, then click Proceed. Next you get an error that the publisher doesn’t trust the UA Web Client. To resolve this error, copy the UA Web Client certificate from the "D:\docker\Rejected Certificates\certs" folder to the "D:\docker\UA Applications\certs" folder on the gateway. You do not need to restart of the gateway. Repeat this step. You can now connect to the gateway from the cloud, and you are ready to add OPC UA servers to the solution.

We simply get an error from the connected factory web site

A clarification we have - we are running the publisher and proxy modules (pulled from docker) but it is unclear if we need to have them function inside an iot-edge instance proper?

barnstee, 2017-07-19:
Running the Docker containers from DockerHub is sufficient. Most folks have problems with the virtual Docker network and doing name resolution to physical networks to connect to their OPC UA servers. We're working on a solution right now to make this way simpler. In the mean time, you can try this workaround: Add a hosts file entry for each OPC UA server you want to connect to on the machine hosting the Docker containers.

PedroFloresGuillen, 2017-07-19:
KZeronimo, did you check if the connected factory and the proxy are from the same version?
When trying to connect it should appear in the proxy that the version was not correct.
If this isn't the problem, I won't know how to help you.

KZeronimo, 2017-07-19:
For versions - I believe we are aligned with what the ConnectedFactory created from iotsuite expects

- publisher v1.1.1

- proxy v0.1.4.1

@barnstee - We've taken care adding the the opcua svr to the local host file - the net effect is that we can ping the opcua svr from the publisher docker container and it appears that the publisher starts up successfully (note I'm providing a publishednodes.json)
[image]

We still get an error when trying to connect to opc.tcp://publisher.meshfactory.com:62222

KZeronimo, 2017-07-19:
Update - we are now able to establish a connection from Connected Factory to `opc.tcp://publisher.kz.meshiot.io:62222` - public certs are swapped as expected and I can browse the publisher - note the domain we just happened to pick prior _meshfactory.com_ was a domain that was registered on the web - even though we had placed records in the local hosts files it seems that the something was getting clobbered when the proxy was trying to get address info for the publisher - so changing the app name from meshfactory.com to meshiot.io allowed progress

Now I'm trying to connect the actual opcua server `opc.tcp://opcuasvr1.kz.meshiot.io:48010` to the Connected Factory - I have manually installed/trusted the UA Web Client into this server - I get in a loop when telling Connected Factory to trust the servers self signed cert

[image]

KZeronimo, 2017-07-19:
Last update - all is now working - documenting in case this helps someone

We made sure that the SAN on the opcua svr cert **only** contained `opcsvr1.kz.meshiot.io` - I distributed this new server cert to the publisher and UAExpert - all connected locally successfully

Now we can connect, browse, publish, etc on opc.tcp://opcuasvr1.kz.meshiot.io:48010 from Connected Factory

Finally, I did make sure the opcua svr was assigned an IP address from the docker network and put a record in my Windows hosts file

My Windows hosts file contains

```
10.0.75.1			opcuasvr1.kz.meshiot.io
10.0.75.2			publisher.kz.meshiot.io
```

Feel  free to close this issue if there is no other discussion or if anything needs corrected

barnstee, 2017-07-20:
Well done folks. I know this is painful right now and we will make this better.
This part is interesting:

note the domain we just happened to pick prior meshfactory.com was a domain that was registered on the web - even though we had placed records in the local hosts files it seems that the something was getting clobbered when the proxy was trying to get address info for the publisher - so changing the app name from meshfactory.com to meshiot.io allowed progress

@marcschier: Another reason we need the whitelisting feature, or at least limit Proxy to search on the local subnet for the server!

closed 2017-07-20 by barnstee

ryan-ms, 2018-07-10:
Hi @KZeronimo ,

I had same problem with you. But after looking your information, It' still not worked  successfully.

I can now connect to the publisher from connected factory portal, but I still stuck in the loop of telling Connected Factory to trust the servers self signed cert.

Can you explain more detail how you slove it?

========================

My environment as below:

OPCUA Server: Real OPCUA Server (Private IP: 192.168.2.150)
OPCUA Gateway: My laptop (Private IP: 192.168.2.40)

And on the Gateway, I add the following 2 entries in host file.

192.168.2.40       opcpublisher.ryan.io
192.168.2.150     opcserver.ryan.io 

My steps:

1.	Run publisher on docker:

docker run --rm -it -v C:\cfShared:/docker -v x509certstores:/root/.dotnet/corefx/cryptography/x509stores --network iot_edge --name publisher -h publisher -p 62222:62222 --add-host OPC-A:192.168.2.150 microsoft/iot-edge-opc-publisher:2.1.3 publisher "HostName=opcuaicpcf507d3.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=iry8pn5zWkXXkYar+sicjCcqqgUuxvPZmyF2Yk678M8=" --lf /docker/publisher.log.txt --as true --si 1 --ms 0 --tm true --vc true --di 30

2.	Run Proxy on docker:

docker run -it --rm -v C:\cfShared:/mapped --network iot_edge --name proxy --add-host OPC-A:192.168.2.150 microsoft/iot-edge-opc-proxy:1.0.2 -i -c "HostName=opcuaicpcf507d3.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=iry8pn5zWkXXkYar+sicjCcqqgUuxvPZmyF2Yk678M8=" -D /mapped/cs.db

and

docker run -it --rm -v C:\cfShared:/mapped --network iot_edge --name proxy --add-host OPC-A:192.168.2.150 microsoft/iot-edge-opc-proxy:1.0.2 -D /mapped/cs.db

3.	Copy the publisher certification file and UA Web Client certificate (Download from Connected Factory) into trusted folder in OPCUA Server.

4.	Copy UA Web Client certificate (Download from Connected Factory) into trusted folder in OPCUA gateway.

5.	Enable gateway :  enter opc.tcp://publisher:62222 in the page of “connect to your opcua server” and click “process” after the certificate warning, and success.
So we can observe the publisher on dashboard.
 

6.	Connect to OPCUA Server (Failed)
Disconnected the publisher connection on dashboard, and enter opc.tcp://opcserver.ryan.io:51210, and click connect.
And it will keep asking to trust certificate, after click the “process”

[image]

 
Can you give me some instruction? 
Thank you!