Azure/iot-edge-opc-publisher-v1 #59: Certificates are lost after every restart of Docker (at least on Windows Server 2016)

robert4os, 2017-07-19:
Hi, 
in the Readme it says
> On first run, for one-time IoT Hub registration:
> docker run -it --rm gw <applicationName> <IoTHubOwnerConnectionString>
> From then on:
> docker run -it --rm gw <applicationName>

1) However the Certificate Store (new X509Store("IoTHub", StoreLocation.CurrentUser)) does not persist from one launch to another. 
- Obvious problem: connection string has to be passed every time.

2) Nor is the Application certificate which is stored in "CurrentUser\\UA_MachineDefault").
- Thus the publisher uses a new Certificate at every launch. This is a problem, because our OPC UA Server has to be configured to trust it after every docker restart. 

best regards, Robert

barnstee, 2017-07-19:
Please follow the link at the bottom of the Docker instructions for how to persist the log, cert and publishednodes.json:
"For detailed instructions on using Docker with the OPC Publisher, see here."

barnstee, 2017-07-19:
Important: Please replace the path '''/build/src/GatewayApp.NetCore/bin/Debug/netcoreapp1.0/publish''' with '''/build/out'''

robert4os, 2017-07-19:
ok I see, it's more complicated ;)

barnstee, 2017-07-19:
Isn't it always. :-)

robert4os, 2017-07-19:
Unfortunately the documentation at said link does not make it much easier ;) There seems to be some mixup between Linux and Windows paths.
In addition I have to use Docker Toolbox, so I can only share within C:\Users.

I managed to place the 'CertificateStores' folder in the shared folder.  
docker run -it --rm -h OPCUAPublisherApp -v c:\Users\docker:c:/build/out/CertificateStores gw OPCUAPublisherApp

But system certificates are not persisted. I suppose it has to do with '/root/.dotnet/corefx/cryptography/x509stores'...  where/what should this be?

Thank you

barnstee, 2017-07-20:
This is the OPC UA application certificate and private key store location. We use the X509 store class for this, as it provides more secure storage than the file system (on Windows, on Linux it still uses the file system on this path /root/.dotnet/corefx/cryptography/x509stores).
But the drawback is that you can't map the folder to your host machine, which means that on every startup, a new cert is created. :-(
So the work around this, we need to switch back to directory-based certificate stores. This can be done in the ModuleConfiguration.cs where the StoreType is set to "X509Store". Switch it to "Directory" and set the path to "CertificateStores/UA Applications". I'll make the change in the Publisher.

barnstee, 2017-07-20:
Actually: One thing to clarify: Are you using Linux containers or Windows containers?

robert4os, 2017-07-20:
Thank you for your help!

Yes, I actually played with this switch to 'Directory' workaround - it works. 

There is also the SecureIoTHubToken which also uses the X509Store, so that the connection-string will not be remembered. So this would have to switch to 'Directory' as well. 

Maybe it would be useful to have a switch to either use X509Store or Directory throughout the entire application. Could be an environment setting. 

I am using Windows containers (have to).

robert4os, 2017-07-20:
Btw, the SecureIoTHubToken will have to have another name when stored in 'Directory' to avoid collision with the ApplicationCertificate - or to reduce confusion, if it is stored in another directory 'store' :)

hansgschossmann, 2017-08-22:
I will address parts of this in my upcoming pull request, which gives you the option to control all store types and path values via command line. Maybe you could spend a few cycles and test this when it is merged.

hansgschossmann, 2017-11-14:
@robert4os: fixed a while back in 2.1.0 already

closed 2017-11-14 by hansgschossmann