Azure/iot-edge-opc-publisher-v1 #61: Isn't the storage of ApplicationCertificate in Directory redundant?

robert4os, 2017-07-20:
In ModuleConfiguration.cs the ApplicationCertificate is stored in X509Store, which is fine.

But later in ModuleConfiguration.cs below "// Ensure it is trusted" it will be stored again in 'Directory' store for TrustedPeerCertificates. 

I do not understand, why this step is necessary. It certainly works without it. 

When switching the ApplicationCertificate store to 'Directory' (see https://github.com/Azure/iot-edge-opc-publisher/issues/59 ) then the code will stumble, because a certificate with the same thumbprint already exists.

Maybe it would help clarify if there would be clearly separated certifcate stores, like so: 'UA Clients' and 'UA Servers'. (and 'UA Rejected Servers')?

thank you and best regards, Robert

barnstee, 2017-08-01:
Yeah, I think this is a bug. When a server creates its cert, it is automatically added to the trusted store, so this is redundant. I also agree that the naming of the stores in confusing.

hansgschossmann, 2017-08-21:
For certain scenarios this might make sense. The intention is to allow different OPC applications on the same host allow access to each other. For a scenario like the simulation used in Connected Factory (see https://github.com/Azure/azure-iot-connected-factory) this allows OPC servers and clients which are running in a container to share the same trusted cert directory and trusting each others. 
I will make the storage of the publisher server cert in the trusted directory configurable via command line in one of the next pull requests.
The naming i do not like neither. LDS-ME is using 'issuer', 'own', 'trusted', 'rejected' as directory structure, which i like much more. This is something which may need some coordination with the .NET stack folks, but from my point of view it is worth doing it. I will open a feature request against the stack.

hansgschossmann, 2017-11-14:
@robert4os: closing this one. if you still see issues with the latest version, please open a new issue

closed 2017-11-14 by hansgschossmann