Azure/iot-edge-opc-publisher-v1 #63: Orphaned messages are received in IoTHub

JayeshThamke, 2017-08-02:
In IoTHub, I have started receiving messages to IoTHub via AMQP telemetry and UaJSON format. However, I have found these messages are Orphaned Messages and hence I cannot create a Blob Storage using Stream Analytics job.

Did anybody came cross this issue earlier?
Is there any configuration items or properties need to be changed?

Thanks for reply,
Jayesh

[image]

JayeshThamke, 2017-08-06:
I have corrected message routing and it works.

closed 2017-08-06 by JayeshThamke

marwaadihu, 2019-06-14:
Hey Jayesh, Can you please tell what you corrected in message routing, in my case it is correct routing as per requirement, although I am getting this issue