Azure/iot-edge-opc-publisher-v1 #70: cannot connect to IoT Hub with mqtt

lcao8, 2017-08-17:
I use docker to run the publisher and it times out connecting the IoT Hub. If I change transport protocol from mqtt to http, it connects fine.

 docker run -it --network=host --rm -v publisher:/build/out/Logs -v publisher:/build/out/CertificateStores -v publisher:/shared -v publisher:/root/.dotnet/corefx/cryptography/x509stores -e _GW_PNFP="/shared/publishednodes.json" gw publisher.st.com "my-connection-string"
08/17/2017 19:45:49: Publisher is starting up...
08/17/2017 19:45:49: Adding certificate to trusted peer store. StorePath=CertificateStores/UA Applications
08/17/2017 19:45:49: A certificate with the same thumbprint is already in the store.
08/17/2017 19:45:49: Could not add certificate to trusted peer store. StorePath=CertificateStores/UA Applications
08/17/2017 19:45:49: Starting server on endpoint opc.tcp://publisher.st.com:62222/UA/Publisher...
08/17/2017 19:45:50: Server started.
08/17/2017 19:45:50: Attemping to register ourselves with IoT Hub using owner connection string: 
08/17/2017 19:45:52: Attemping to read connection string from secure store with certificate name: publisher.st.com
08/17/2017 19:45:52: Attemping to configure publisher with connection string: 
08/17/2017 19:46:22: One or more errors occurred. (One or more errors occurred. (connection timed out: 40.76.71.185:8883))
08/17/2017 19:46:22: Unhandled exception in Publisher, exiting!

khilscher, 2017-08-17:
Is your firewall blocking MQTT port 8883 outbound? Many firewalls will block this port while allowing HTTPS.

lcao8, 2017-08-18:
Thanks! it's likely the cause. Is there a way to select the transport at run time?

khilscher, 2017-08-18:
Suppose you could wrap the MQTT connection attempt in a try/catch, then revert to HTTPS. Or add a command-line argument.

hansgschossmann, 2017-08-21:
this is a problem a lot of folks will hit. i will prepare a pull request which will add command line parameter and option handling to the publisher.

hansgschossmann, 2017-08-26:
@lcao8 look for CreateFromConnectionString and change the Mqtt parameter to Http1
You could also check out what is checked in my fork under https://github.com/hansgschossmann/iot-edge-opc-publisher, where you could specify the protocol via command line. I will create a pull request when i am back from vacation.

hansgschossmann, 2017-11-14:
@lcao8: v2.1.0 could be configured via command line to use any transport protocol supported by the Azure IoTHub Client SDK.

closed 2017-11-14 by hansgschossmann