Azure/iot-edge-opc-publisher-v1 #71: Cannot reconnect to OPCUA server

zhironghsu, 2017-08-21:
Hi,

I run the publisher and get the following message when the OPCUA server is down, then I restart the OPCUA server, the publisher cannot reconnect to the server.

Status: BadNoCommunication 'Server not responding to keep alive requests.'/t/tOutstanding requests: 0/t/tDefunct requests
: 0

hansgschossmann, 2017-08-21:
Interesting, I will give it also a try over the next few days.

carlwouters, 2017-08-25:
Just updated to latest version and getting similar errors from time to time which causes app to shutdown. Error message in attachment.
[error.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1251825/error.txt)

Shouldn't the app need to try to reconnect?

hansgschossmann, 2017-08-25:
yeah, i will send a pull request out after i return from vacation on the 9/4.
if one would like you folks could check out my for which has all the changes at https://github.com/hansgschossmann/iot-edge-opc-publisher.git

zhironghsu, 2017-08-28:
There is batching message in the code. 
Why not use DeviceClient.SendEventBatchAsync() directly? If I am misunderstanding something?

hansgschossmann, 2017-09-03:
The implementation we have in the code base is independent from the transport. SendEventBatchAsync is right now only implemented in Http. Mqtt for example just sends each message as a single message to IoTHub. Hope this answers your question.

hansgschossmann, 2017-11-14:
@zhironghsu: v2.1.0 implements a different server connection strategy. please open a new issue if you still see problems.

closed 2017-11-14 by hansgschossmann