Azure/iot-edge-opc-publisher-v1 #72: Make sampling interval configurable for each node

carlwouters, 2017-08-23:
It would be ideal if the sampling interval can be configured for each node separately within the publishednodes.json

hansgschossmann, 2017-08-25:
i thought about this as well. most likely i will look into that before i submit my PR. it would make sense to change the default sampling rate of 1000ms via cmd line param and also allow to change on a node level. which scenarios do you have in mind, which need a change? slowing it down or increasing it or both? i ask because this may interfere with the batching functionality i merged in yesterday.

carlwouters, 2017-08-25:
In our case the 1s is perfect as default and would be nice to change through cmd, there are just certain sensor values which we want to set to 60s or something as some of them don't rapidly change. (Ambient temperatures, pressure, humidity, ...).

I guess a nice solution would be to group them. My feeling would be that there won't be alot of applications which will require to define a separate interval for each node.

Thanks alot by the way, that batch mode works smoothly and efficiently.

carlwouters, 2017-09-25:
@hansgschossmann I've been testing out your last commits.
Changed all our OPC UA nodes in 2 sections (in the publishednodes file):
1) each warnings, run cmds, etc.. in a section of 1sec publish interval
2) sensor data to 30sec publish interval

But it seems there are queing problems?
The source timestamps are creeping away from the published timestamps as can be seen in the attachment.

[image]

closed 2017-10-10 by carlwouters