Azure/iot-edge-opc-publisher-v1 #74: Running the module without azure

Zuendelmeister, 2017-09-06:
Hi you guys,
I'd like to get the gathered information just to the iod-edge broker and not to the cloud.
Is there a way to do this with your module?

Thanks
Best regards,
Zuendelmeister

hansgschossmann, 2017-09-07:
sure. just checkout how we send it to iothub. you could put your code in there and do whatever you want to do with the data. what is the scenario you are going to address?

Zuendelmeister, 2017-09-11:
I just want to get the data from the OPC-OA server and send them to the iot edge broker. Before loading them to iot hub I need to add some additional information. Is there a plan to add this feature?

Best regards,
Zuendelmeister

hansgschossmann, 2017-09-11:
Not yet. I keep the issue open and when we update the publisher to the iot-edge release, which will happen later this year, we will look into this.

Zuendelmeister, 2017-09-12:
Thank you so much =)

sebader, 2018-06-07:
When running the publisher as an IoT Edge module, you just need to configure your [Edge routes ](https://docs.microsoft.com/en-us/azure/iot-edge/module-composition#declare-routes)appropriately. This way you can avoid to send any data to the IoT Hub in the cloud. For instance, you could route all the output from the publisher module into another Edge module that sends the data into some on-prem storage.

closed 2019-04-30 by markusstuhler