Azure/iot-edge-opc-publisher-v1 #77: Browsing every nodeId in my device

Nsquare1994, 2017-09-21:
Hi guys,
I was wondering if it is possible for me to search for all nodes names, ids, values and etc from the device that I'm using and implement it in program.cs.

Thank you for your time.
Best regards,
Eric

hansgschossmann, 2017-09-24:
Sure you could do. Publisher does use the OPC UA client interface to create sessions, subscriptions and monitored items. You could easily use the OPC UA stacks Browse API to walk through the address space.
I am not sure though how much sense it makes, since a lot of the nodes have constant values. Also you may implement it to only publish variables and exclude methods and other nodes, which make no sense. 
Once i also thought about implementing this as a built in feature in a way to be able to configure a root of a tree of nodes to publish in your case this would be the root node. But as i said i have not got more requests on this and filtering out nodes which make no sense is some effort.

closed 2017-09-24 by hansgschossmann

Nsquare1994, 2017-09-29:
Dear sir,
Thanks for the advice, but I'm having trouble finding the api that you're referring to. If you have the spare time, would it be possible for you to give me a little tip on that.
Thanks for the help, hope you have a great day.

Best regards,
Eric