Azure/iot-edge-opc-publisher-v1 #79: .NETCore 2.0 (ARM Support)

MSIoT, 2017-09-25:
My partner would like to run Publisher on ARM-based gateway, and looking for .NET Core 2.0 support.

JuergenKosel, 2017-09-28:
Hello @MSIoT ,

I have the same problem, that I need the publisher for an arm device.
But anybody I have asked, told me, that the .cs based publisher could not be build for arm-devices.
So I see only the solution to rewrite the publisher from scratch in C/C++. By using the Azure SDK from https://github.com/Azure/azure-iot-sdk-c

Greetings
  Juergen

hansgschossmann, 2017-10-11:
pls check out the release 2.1.0 which is available now and supports .NET Core 2.0

The .NET Core 2.0 docker images are available for [ARM32 and ARM64](https://hub.docker.com/r/microsoft/dotnet/). Please check other download sites for the installable version.

closed 2017-10-11 by hansgschossmann