Azure/iot-edge-opc-publisher-v1 #82: Data timestamps running away

carlwouters, 2017-10-10:
We have been testing the latest commits.
Our nodes are in the format:

{
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=1",
                "OpcSamplingInterval": 1000
}
{
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=2",
                "OpcSamplingInterval": 1000
}
{
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=3",
                "OpcSamplingInterval": 30000
}
{
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=4",
                "OpcSamplingInterval": 30000
}

All sensor data is sampled at 30s while run cmds etc.. at 1s. (In total more than 300 data points at the moment)
We see that the actual timestamp of the sensor data is creeping away from the timestamp IoT Hub receives the messages:

[image]

Are we missing something?

hansgschossmann, 2017-10-23:
i will look into this in more detail in the next weeks. since we support now batching of the messages to IoTHub, could you first do a restest by disabling batching via commandline parameters. The iothub-enqueue timestamp includes also network latency and is non determenistic. The order of the messages should be preserved at least.

carlwouters, 2017-10-23:
How can I disable batching?
I've put --ms at 0, but no difference.

Maybe as additional information, it seems the edge application is only publishing at 30s, but due to the amount of nodes, it can't publish them all together so it queue's them up for next publish interval?

[image]

hansgschossmann, 2017-10-23:
with --ms 0 it will send each 1 sec interval. with an additional --si 0, it will send each message. i will look into that in the next few weeks.

carlwouters, 2017-10-23:
It seems that because I put sampling interval at 30s, it will only send messages to Hub each 30s even though it can't send all messages in one time.

hansgschossmann, 2017-11-14:
hi carl, please update the issue with your latest findings and let me know if there are still issues.

carlwouters, 2017-11-14:
Solution:
- Update namespace tags (ns to nsu)
- Update to version 2.1.1 solved all latency issues

Everything is now running smoothly.

closed 2017-11-14 by carlwouters