Azure/iot-edge-opc-publisher-v1 #83: Error "Predefined type 'System.xxxxxxx' is not defined or imported"

qwertyzq, 2017-10-17:
I am error when i download your code and run it in visual studio
it say have error around 1xxx error like this

"Predefined type 'System.Boolean' is not defined or imported"
"Predefined type 'System.Byte' is not defined or imported"
"Predefined type 'System.Double' is not defined or imported"
"Predefined type 'System.xxxxxxx' is not defined or imported"
Etc. 

I don't know why? how can i fix it?
thank you and best regards.

barnstee, 2017-10-17:
Please verify that you have installed the .NetCore 2.0 SDK.

carlwouters, 2017-10-17:
Also take care you have the latest VS update to take care .NetCore 2.0 works.

qwertyzq, 2017-10-18:
I already install .NetCore 2.0 SDK From [https://www.microsoft.com/net/download/core](url) 

but it still error.

qwertyzq, 2017-10-18:
I already install .NetCore 2.0 SDK from [https://www.microsoft.com/net/download/core#/sdk](url)
but it still error.

hansgschossmann, 2017-10-23:
Have you tried to run a dotnet restore src/OpcPublisher.sln? You need to ensure that your nuget is configured correctly.

hansgschossmann, 2017-11-14:
@qwertyzq: have nothing heard back for a while. closing it now. please open a new one if you still see issues.

closed 2017-11-14 by hansgschossmann