Azure/iot-edge-opc-publisher-v1 #86: publisher could be accidently terminated

JuergenKosel, 2017-10-26:
Hello,

for productive usage the publisher is expected to run permanently.
But the publisher reads from standard input:
https://github.com/hansgschossmann/iot-edge-opc-publisher/blob/a43e00e6a6b2d3d23fe884981ec734380acf3b12/src/Program.cs#L479
So the publisher terminates, if someone gives accidentally a key stroke to the terminal running the publisher.
If trying to start the publisher in some daemon like mode from an init script, it terminates immediately after initialization due to the lack of standard input.

hansgschossmann, 2017-11-05:
I see your point. A "detached terminal" mode would make sense to add.

barnstee, 2017-11-06:
The NetCoreConsoleServer implements this, have a look at:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/SampleApplications/Samples/NetCoreConsoleServer/Program.cs

hansgschossmann, 2017-11-14:
i will address this in the next release

JuergenKosel, 2017-11-15:
Fixed in https://github.com/Azure/iot-edge-opc-publisher/tree/2.1.2

closed 2017-11-15 by JuergenKosel