Azure/iot-edge-opc-publisher-v1 #87: Alert if internal queue almost full

khilscher, 2017-10-26:
Log an alert if the internal queue (_monitoredItemsDataQueue) is close to full (BoundedCapacity) so a monitoring system, such as OMS, can notify an administrator.

hansgschossmann, 2017-11-14:
i need to look into this in more detail. sending a message to IoTHubs operation management, would definitly make sense.

hansgschossmann, 2017-11-14:
accidently pressed the close button:(

khilscher, 2017-11-15:
No worries Hans ;-) Just an FYI, Operations Monitoring on Azure IoT Hub is being deprecated. See https://azure.microsoft.com/en-us/blog/monitor-your-azure-iot-solutions-with-azure-monitor-and-azure-resource-health/

Some other possible options:
1. Expose a direct method which returns diagnostic data from the gateway (basically the data you already log with --diagnosticsinterval). Customers can write an Azure Function, for example, that can periodically call the method and can then act on the returned information. For example, generate alert.
2. Use device twins - customer's monitoring system must periodically poll device twin data
3. Periodically send diagnostic data in JSON format to the IoT Hub as a D2C message. Message routing would need to be enabled in IoT Hub to route the message to another endpoint for processing. 
4. Log the alert message to the log file (if diagnosticsinterval is enabled) and put the onus on monitoring for the alert on the customer using a log reader agent.