Azure/iot-edge-opc-publisher-v1 #89: BadCertificateUntrusted although autotrust option is on

toddysm, 2017-11-06:
Hello,

We are seeing the following error although the `--autotrustservercerts=` option is set to `True`:

```11/06/2017 02:23:49.001 Creating task process and batch monitored item data updates...
11/06/2017 02:23:49.911 Connect and monitor session and nodes on endpoint 'opc.tcp://192.168.1.199:4840/'.
11/06/2017 02:23:51.326 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.03.340.0UA-TCP 1.03.340.0] [ID=59001] Connected To: opc.tcp://192.168.1.199:4840/ [None/None/Binary]
11/06/2017 02:23:51.642 Create session for endpoint URI 'opc.tcp://192.168.1.199:4840/' with timeout of 10000 ms.
11/06/2017 02:23:51.720 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.03.340.0UA-TCP 1.03.340.0] [ID=59002] Connected To: opc.tcp://192.168.1.199:4840/ [None/None/Binary]
11/06/2017 02:23:52.271 ***EXCEPTION*** BadCertificateUntrusted 'Certificate is not trusted.
SubjectName: CN=XXXXXX, O=XXXXXX, L=XXXXXX, S=XXXXXX, C=XXXXXX, DC=XXXXXX
IssuerName: CN=XXXXXX, O=XXXXXX, L=XXXXXX, S=XXXXXX, C=XXXXXX, DC=XXXXXX' Certificate is not trusted.
SubjectName: CN=XXXXXX, O=XXXXXX, L=XXXXXX, S=XXXXXX, C=XXXXXX, DC=XXXXXX
IssuerName: CN=XXXXXX, O=XXXXXX, L=XXXXXX, S=XXXXXX, C=XXXXXX, DC=XXXXXX
11/06/2017 02:23:52.619 Certificate 'CN=XXXXXX, O=XXXXXX, L=XXXXXX, S=XXXXXX, C=XXXXXX, DC=XXXXXX' will be trusted, since the autotrustservercerts options was specified.
11/06/2017 02:23:52.782 Channel 0: ProcessErrorMessage(BadCertificateInvalid 'Error received from remote host: ')
11/06/2017 02:23:52.786 ***EXCEPTION*** BadCertificateInvalid 'Error received from remote host: '
11/06/2017 02:23:52.799 ***EXCEPTION*** BadCertificateInvalid 'Error establishing a connection: Error received from remote host: '
11/06/2017 02:23:53.145 Session creation to endpoint 'opc.tcp://192.168.1.199:4840/' failed 1 time(s). Please verify if server is up and Publisher configuration is correct. BadCertificateInvalid 'Error establishing a connection: Error received from remote host: '

========================================
Id: BadCertificateInvalid
Description: Error establishing a connection: Error received from remote host: 
>>> Error establishing a connection: Error received from remote host: 
---    at Opc.Ua.Client.Session.<Create>d__72.MoveNext()
--- --- End of stack trace from previous location where exception was thrown ---
---    at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
---    at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
---    at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
---    at OpcPublisher.OpcSession.<ConnectSession>d__22.MoveNext() in C:\Users\toddysm\OneDrive\Development\Projects\Remote\iot-edge-opc-publisher\src\OpcSession.cs:line 373
===
Id: BadCertificateInvalid
Description: Error received from remote host: 
>>> Error received from remote host: 
---    at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
---    at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.OnConnectOnDemandComplete(Object state)
========================================```

Our assumption is that this issue is showing up because the OPC-UA server we are trying to connect doesn't trust the Publisher (or is that assumption wrong?) but not sure how to resolve this.

Also, we don't see any certificates in the rejected certificates folder.

barnstee, 2017-11-06:
Righty on the money: The error comes from the remote server. You need to trust the Publisher cert in your remote server somehow.

toddysm, 2017-11-07:
Gret! Thanks for confirming my theory :) Now, of course, the next question is: How do I know which one is the Publisher certificate? In the trusted certificates folder I see 3 certificates that have the name
`[application_name] [fingerprint].der`
One of those, I assume is the Publisher but not sure which one. The second one must be the server I want to connect to. The third one, I think showed up after I restarted the Publisher, which makes me think the Publisher cert gets dynamically generated. Any suggestions?

toddysm, 2017-11-07:
After some more digging, it doesn't seem I will be able to upload the certificate to the device because it is vendor provided device and we don't have access to the underlying OS. We were able to connect to the device using UAExpert and are able to read information from it via OPC-UA without configuring any certs on the device - is there a way to achieve the same configuration with the Publisher?

barnstee, 2017-11-07:
Are you running the Publisher in a Docker container? If you don't map the certificate folder of the Docker container to the host, yes, the cert will be re-created at each start.
It is possible that within UA Expert, you disabled security and therefore were able to connect to he server. This is not recommended for obvious reasons.
Does the server come with a configuration tool? Usually within this tool, there is an option to trust client certificates (like the Publisher one).

toddysm, 2017-11-11:
@barnstee The server doesn't support secure communication at all. We had to turn it off from the code to make it work.

barnstee, 2017-11-11:
I doubt that. What's the server's make and model?

hansgschossmann, 2017-11-14:
@toddysm: i will support a per endpoint configuration option to allow also unsecure communication to a endpoint. a preview is available at https://github.com/hansgschossmann/iot-edge-opc-publisher

hansgschossmann, 2017-11-17:
this is now available with commit 60c558f50da63bf48d78225cbbdeee4abd61eff5

toddysm, 2017-11-18:
Thanks @hansgschossmann - I've already tested the configuration and it looks good. Thanks for the changes!

barnstee, 2017-11-18:
Folks, I'm not happy with the outcome here: Turing off security should be the LAST RESORT. Just because it is not obvious how to turn on security on the server is not a valid reason to use unsecure communication. What make and model is your OPC Server? Unless it is using a Nano profile, it must support security.

hansgschossmann, 2017-11-18:
This is not an issue in OpcPublisher. Please follow up with @toddysm offline to get this sorted out.

barnstee, 2017-11-18:
Please leave it open so we can get to the bottom of this. I don't want to set the precedence that you just turn off security and all is fine!

toddysm, 2017-11-18:
@barnstee As discussed offline - this is a hardware device that supports OPC-UA protocol that we would like to connect to Azure IoT hub. The vendor doesn't have immediate plans for supporting secure communication to the device. We needed to turn off security for Proof of Concept and to know what features need to be implemented in the next version of the device software

barnstee, 2017-11-19:
OK, that's fine, that's the right approach!

closed 2017-11-19 by barnstee