Azure/iot-edge-opc-publisher-v1 #90: Json Message sent to IoTHub is not valid

Banchio, 2017-11-07:
It seems that when constructing a json message with several monitoredItems, the outputted json is invalid. It miss the square brackets (first and last characters) and commas between the monitoredItems. Json message was valid with IoTHubMessaging.cs before commit https://github.com/Azure/iot-edge-opc-publisher/commit/194f30c8b81d3c16f32a8aa3be93e937aae5278f#diff-51136702e85626f290d9396364799eda so probably a regression was introduced with the monitoredItemQueue. Thanks

hansgschossmann, 2017-11-14:
@banchio: yes, this is too bad. i have a fix in the works with more options to configure the telemetry which is ingested. pls check out the preview in https://github.com/hansgschossmann/iot-edge-opc-publisher
the fix will ingest always an array even there is only one single message and the data is consumable by JSON deserializers right away.

hansgschossmann, 2017-11-17:
this is fixed with 60c558f50da63bf48d78225cbbdeee4abd61eff5

closed 2017-11-17 by hansgschossmann