Azure/iot-edge-opc-publisher-v1 #91: Unable to decode Extension Object

hkmardon, 2017-11-09:
I am subscribing to a node that contains an extension object, however this does not get decoded and comes through as follows: (I've removed the server & node id items)

"Value":{"Value":{"TypeId":{"Id":"i=5004"},"Body":"LYKyvfRY0wEBAABRCgAAAAAAAAAAAAAAAAAAAAAAAA=="}

In reality the Value contains several member items which I can view successfully using the UAExpert client tool.

Are extension Objects with custom datatypes supported? If so how are they decoded so their member values can be read?

JuergenKosel, 2017-11-10:
Hello,
this looks base64 encoded, on Linux (WLS, cygwin,...) try:
```
echo "LYKyvfRY0wEBAABRCgAAAAAAAAAAAAAAAAAAAAAAAA==" | base64 -d > out.hex
hexdump out.hex 
0000000 822d bdb2 58f4 01d3 0001 5100 000a 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
000001f
```

gasherma, 2018-06-07:
No feedback assuming Juergens comments resolved the issue.

closed 2018-06-07 by gasherma