Azure/iot-edge-opc-publisher-v1 #95: docker run iot-edge-opc-publisher failed with error "shuold specify access keyName"

chp008, 2017-11-17:
Question One:
Hi, I want to start the opc-publisher with the following step:
1: build an image with tag localhost/opcuapublisher
2: run as follows:
docker run -p 62222:62222 microsoft/iot-edge-opc-publisher localhost/opcuapublisher  
HostName=xxxxxx.azure-devices.cn;SharedAccessKeyName=iothubowner;SharedAccessKey=xxxxxxxxP4QwEIpokbvIGcxE=

but it failed with error:
Should specify accessKeyName
Error in InitAsync(Should specify accessKeyName)

Question 2:
since the default publishednode.json is not be correct by default. we shuold make a correct confiugration. Do we have an parameter in docker to volumen another file for it.

hansgschossmann, 2017-11-17:
answer to Q1:
have you tried using quotes around the connection string? 
localhost/opcuapublisher is not a good name for the application. i know this is missing in the usage. it should be a valid DNS hostname label.

answer to Q2:
there is no way making it correct, since this depends on the OPC UA server to connect to. the file is supposed not to be in the docker container, but in the docker hosts file system. use a -v and specify a bind mount like pointed out in "Using bind mounts (shared filesystem)". Maybe i should add a sample. On my Win10 dev host i use "-v //d/docker/docker" and my publishednodes.json is in d:\docker\docker.

answer to Q3:
Azure services are protected against DoS attacks. Have you tried to connect to your hub using DeviceExplorer (https://github.com/Azure/azure-iot-sdk-csharp/tree/master/tools/DeviceExplorer) or iothub-explorer (https://www.npmjs.com/package/iothub-explorer)?

gasherma, 2018-06-07:
Hans addressed the questions above

closed 2018-06-07 by gasherma

Norman03, 2018-08-01:
Hi Gasherma,

I used below command and even I tried the solutions posted by @hansgschossmann 
Could you please help me on the same?

`sudo docker build -f Dockerfile -t elastic_villani . && sudo docker run -v /home/sysadmin/iot-edge-opc-publisher/opcpublisher:/appdata -t elastic_villani device_rpi_testing HostName=xxxxx.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=QkMajksdghfjas+/qrxBAKi2Lqb/tUZU=`

Norman03, 2018-08-02:
Issue is fixed. Thanks!