Azure/iot-edge-opc-publisher-v1 #100: Puplisher 2.1.3 can not subscribe from some OPC UA servers anymore

JuergenKosel, 2017-11-27:
Hello,

I have recently updated the opc publisher from version 2.1.2 to version 2.1.3
Now the opc publisher refuses to publish variables from some opc servers:
```
11/27/2017 09:57:37: Object reference not set to an instance of an object.
11/27/2017 09:57:37: Error processing monitored item notification
```
With the previous version, this has worked.

JuergenKosel, 2017-11-27:
With version 2.1.2 the delivered data looks in device explorer like this:

```
27/11/2017 11:51:33> Device: [publisher.example.com], Data:[{"ApplicationUri":"urn:00:06:71:33:00:17:example.com","DisplayName":"ns=3;s=_System.General.Temperature","NodeId":"nsu=urn:System_items;ns=3;s=_System.General.Temperature","Value":{"Value":48.5,"SourceTimestamp":"2017-11-27T10:51:17.7664654Z"}}]Properties:
'content-type': 'application/opcua+uajson'
'devicename': 'publisher.example.com'

```

JuergenKosel, 2017-12-18:
Hello @hansgschossmann 

I have tried with your doker image hansgschossmann/iot-edge-opc-publisher:2_1_3_fix
and it does not solve the problem.:
```
12/18/2017 10:12:32: Object reference not set to an instance of an object.
12/18/2017 10:12:32: Error processing monitored item notification
```

[publishednodes.json.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1567436/publishednodes.json.txt)

hansgschossmann, 2017-12-18:
It was not supposed to fix it, but get more info. Could you also send me the publisher log file?

JuergenKosel, 2017-12-18:
Here is the corresponding publisher logfile:
[publisher.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1568026/publisher.log.txt)

hansgschossmann, 2017-12-18:
hi @JuergenKosel, could you give the docker container another try?

JuergenKosel, 2017-12-19:
With your new docker image it looks _much_ better. It looks like the issue is solved.

hansgschossmann, 2017-12-19:
Could you please send me the publishednodes.json and the publisher.log.txt of this?

JuergenKosel, 2017-12-19:
Yes
[publishednodes.json.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1570799/publishednodes.json.txt)
[publisher.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1570800/publisher.log.txt)

markusstuhler, 2019-04-30:
Closing due to inactivity/solved issue.

closed 2019-04-30 by markusstuhler