Azure/iot-edge-opc-publisher-v1 #101: Can't replace publishednodes.json file

dcrupel, 2018-01-10:
Hi,

I want to replace a broken node in the publishednodes.json file. 

flow:
- execute Docker container in Bash
- edit the node in publishnodes.json (using nano editor) --> publishednodes.json is actually changed
- after a while publishednodes.json is changed again to the version before my changes...

Anyone a solutions?

Thanks!

hansgschossmann, 2018-01-17:
Could you stop the docker container, change the .json and start it again? The .json is only read at startup and then updated by publisher, since there could be new nodes added via the integrated OPC server, which are then persisted in the .json.

dcrupel, 2018-01-24:
Thanks! This works.
Also works (I am using the Wrapper to get from DA to UA): 
- Close Wrapper
- Edit publishednodes.json
- Start Wrapper
- restart IoT Edge (restarts the container)
--> This way I can edit it using cmd line inside the container (sometimes faster for testing)

closed 2018-01-24 by dcrupel