Azure/iot-edge-opc-publisher-v1 #103: Is it possible to monitor child nodes by monitoring the parent nodes?

dcrupel, 2018-01-24:
Since there sometimes are many child nodes underneath a parent node it would be helpful to be able to just monitor the parent node and getting al the data from the child nodes underneath. Is this possible?

Thanks!

hansgschossmann, 2018-01-25:
No. But it sounds like an interesting feature request. I keep the issue open.

damienpontifex, 2018-07-20:
I'm also interested in this. I believe I am looking at the same thing.

e.g. I can monitor "ns=2;s=Channel1.Device1.Tag1" and the data comes through
monitoring "ns=2;s=Channel1.Device1" nothing comes through.

We have a lot of tags under the device, does this currently mean I need to add each one as an entry in the OpcNodes array from the configuration?

I would be happy to contribute a change if you would have any guidance @hansgschossmann ?

mpostol, 2018-07-20:
I have just implemented this feature  in the project:

[Simulator.Boiler](https://github.com/mpostol/OPC-UA-OOI/tree/400/Networking/Simulator.Boiler)

> NOTE the code is on 400 so far. 
This functionality is defined by the interface: `ISemanticDataSource`. If implemented it useing IOC to register recursively all children vars for selected root object.

Unfortunately, it is OPC UA pub/sub but the experience may be useful. 
My concern is that the OPC UA Client doesn't expose appropriate services so it must be implemented on iot edge gateway side.  

Anyway, watch the project or join us at [gitter](https://gitter.im/mpostol/OPC-UA-OOI)

BewaControl-ReneDivossen, 2019-02-15:
@hansgschossmann Has the support for complex data types already been added ?  If yes, where can I find documentation on how to use it?

BewaControl-ReneDivossen, 2019-02-20:
@hansgschossmann Sorry to bother you again with this but can you please give a short update on this matter!  Is there already support for complex data types or will it be added soon?

We are using a GetAlarm module on Siemens SPS and that module saves alarms to an array of complex data types including Id, Message, Timestamp and State.  Right now we have to subscribe to 64 variables just for handling the alarms of the SPS.   There is also the problem of the subvalues of the complex data type not being set at the same time.  Quite frequently we therefore are missing data (e.g. Id and Timestamp are set but State is missing because it hasn't changed yet at sampling time).

hansgschossmann, 2019-02-20:
No, that is still on the backlog.

BewaControl-ReneDivossen, 2019-02-20:
@hansgschossmann OK - than I'll implement a custom module to collect the subitems and package them to a complex object.  Please leave a note in this thread once the feature has been implemented. Thanks.