Azure/iot-edge-opc-publisher-v1 #106: Is it possible to make the publisher send messages, even if nothing is changed?

dcrupel, 2018-02-14:
Hi,

Is it possible to make the publisher send all values instead of only the changed values?

Thanks!

hansgschossmann, 2018-06-05:
The OPC subscriptions monitored items functionality only sends updates when the data value of a node changes. 

I get this request quite a few times. Solution is to implement this programmatically in the publisher, to read the node values at the configured intervals.

So it would be possible, but not yet planned on our side.

dtjensen, 2018-06-06:
I need this same functionality. Can you be a little more specific about where / how we should modify the code?  I am looking at it, but It is not really clear where I should focus for the changes.

BewaControl-ReneDivossen, 2019-02-11:
This would indeed be a nice feature to have!

hansgschossmann, 2019-02-11:
Check out the heartbeat feature added recently. I have not yet created github releases yet but it is available in 2.3.2

Outlook for Android<https://aka.ms/ghei36> herunterladen

BewaControl-ReneDivossen, 2019-02-12:
@hansgschossmann I did not find much documentation about that.  I understood the following:
If I add "--hb=10" to the command line options, it is guaranteed that OpcPublisher sends all subscribed nodes at least once every 10 seconds (and more often im values changed and Sampling and Publishing Interval are set to a value < 10).  Is that correct ?

In the documentation it says that this command line option is used for all nodes that don't have a heartbeat setting of their own.  How would I set a heartbeat for a single node?

jmbrunskill, 2019-02-12:
@BewaControl-ReneDivossen I haven't tried it myself yet, but based on the publishednodes.json example in this repository, you should be able add a key of "HeartBeat" in each node you want to collect. 

```     
"OpcNodes": [
            {
                // identifies the OPC node to publish in either NodeId format (contains "ns=") or ExpandedNodeId format (contains "nsu=")
                "Id": "i=2258",
                // specifies the sampling interval OPC Publisher requests the server to sample the node value
                "OpcSamplingInterval": 2000,
                // specifies the publishing interval OPC Publisher requests the server to publish the node value, it will only be published if the value has changed
                "OpcPublishingInterval": 5000,
                // specifies that there should be a heartbeat generated by OPC Publisher, this means that after the given interval the last message will be sent again with an updated SourceTimestamp value.
                "Heartbeat": 3600,
                // specifies that the first event will not generate a telemetry event, this is useful when publishing a large amount of data to prevent a event flood at startup of OPC Publisher
                "SkipFirst":  true
            }
        ]
```
See: https://github.com/Azure/iot-edge-opc-publisher/blob/master/opcpublisher/publishednodes.json

BewaControl-ReneDivossen, 2019-02-12:
@jmbrunskill Perfect - Thank's a lot!

BewaControl-ReneDivossen, 2019-02-25:
@hansgschossmann I used  **"--hb=20"** to the command line options for a while now and it seems to work as expected.  Today I tried to set a heartbeat value for single nodes and it did not work.
In the **pn.json** I added **"Heartbeat": 1000,** to a single node.  But the publisher kept the global heartbeat of 20 seconds for that node.  I then removed the global heartbeat from the command line options but it still did not work - now there was no heartbeat at all.  I also tried  **"Heartbeat": 10,** (in case it's supposed to be a value in seconds instead of milliseconds as the other intervals), and I tried  **"HeartBeat"** in Camelcase but both did not work either.  
In all the cases I tried, the Heartbeat line was removed from the **pn.json** once the publisher module was stopped and restarted.
Could you please give instructions on how to define heartbeats for single nodes !

hansgschossmann, 2019-02-25:
My bad, it is „HeartbeatInterval” I have fixed the sample and added some more explanation to the readme. So it is:
“HeartbeatInterval”: 20
To achieve the same as you did with your command line option.

BewaControl-ReneDivossen, 2019-02-26:
@hansgschossmann With "HeartbeatInterval" it's working.  Thanks.

closed 2019-04-30 by markusstuhler