Azure/iot-edge-opc-publisher-v1 #110: Publisher stops sending messages after a few days

Jan-Olof, 2018-03-08:
We are using the publisher as a module on Azure IoT Edge. We managed to install it and start it. We can subscribe to OPC UA Servers and get messages. These messages are sent to an IoT Hub and everything works fine.

However after a few days (2-3 mostly) the publisher stops sending messages to the IoT Hub. It still gets messages from OPC UA Servers but it can no longer send them on to the IoT hub. This has happened several times and we have never been able to get the publisher to work more that a few days at a time.

Here is the log file from the publisher from the last time this happened. Unfortunately it had already been truncated when we got to it, so it does not show what happened when the publisher stopped sending messages. Any help would be appreciated. Let me know if more information is needed to find out what has happened.

[publisher.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1792629/publisher.log.txt)

Jan-Olof, 2018-03-09:
Just some additional info on how we get it to work again after a failure. 

If we just stop IoT Edge and start it again it looks like the publisher starts working again. Both the publisher and the hub module logs that messages are now sent to the IoT Hub. However no messages reaches the IoT Hub. 

However, if we remove the publisher module, uninstall IoT Edge, reinstall it, start it and install the publisher again. Then it starts working again, sending messages to the IoT Hub as it should.

varunpuranik, 2018-03-14:
Hi Jan-Olof, 
We will have to investigate this issue. Can you please provide the details of the topology you have? Specifically -
- What modules are deployed on the Edge device? (with some info on the function on each)
- Who is generating the telemetry, and who is consuming it. (or the routes that you have)
- What version of EdgeHub are you running? (should be in the logs at the top)
Also, can you provide the EdgeHub logs when you repro the issue? That could help us as well. 

You have mentioned that this issue could be related to the memory leak issue. Are you seeing any OOM exceptions, or high memory usage on the host (edge device) after a few days?
Thanks,
-Varun

Jan-Olof, 2018-03-16:
Hi,

Here are more details about this issue:

Modules: We have tested with a simple setup with only the publisher and the two standard modules. In these tests we use your publisher from microsoft/iot-edge-opc-publisher:latest. We are running on a Windows 10 1709 machine using Linux Containers.

Telemetry: We are using an OPC UA server in KEPServerEX from which we get the telemetry. In these tests we send the telemetry directly to the EdgeHub module. I will enclose the publishednodes.json.

EdgeHub: Sorry, the top of the logs seem to be truncated on our test machines, and I don't want to restart when doing these tests. Unless this version is less than a week old I assume it is 1.0.0-preview021.10543704. Maybe there should be an easier way to find out what version of a module you are running?

Memory leak: I have not seen any OOM exceptions but I have noticed that the memory usage for the EdgeHub tends to increase over time which is why I thought about it when I saw the memory leak issue.

History: We started running the tests about a month ago. As I mentioned each test failed after 2-3 days. This happened 4 times, after which I wrote this issue.

Current status: On the 9th of March we started a new test run. This time it ran for 5 days without problems. Then we had to restart the tests for other reasons. After this we started tests from 2 different servers with IoT Edge. These tests have ben running for 2 days with no problems. So preliminary results could indicate that the publisher is more stable now. (We will know more next week.) I can see that you have updated the publisher 3 weeks ago. Maybe we are now running this updated version? (I do not know how to check this.)

[publishednodes-telemetryconfiguration.zip](https://github.com/Azure/iot-edge-opc-publisher/files/1819163/publishednodes-telemetryconfiguration.zip)

Jan-Olof, 2018-03-19:
Update: On Saturday the 17th the publisher stopped working again on our Windows 10 machine. Our other test machine (a Windows Server 2016) is still running OK, and has done so for more that 4 days now.

I checked the docker stats on the EdgeHub module. CPU was around 100% (90-105%). Mem usage 514 MB (started at around 80 MB), NET I/O 1,45 GB, BLOCK I/O 357 MB.

The version of EdgeHub is indeed 1.0.0-preview021.10543704.

Here is the publisher log:
[publisher.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1824229/publisher.log.txt)

Just as last time we fail to learn anything from the logs, since they are truncated when we get to them. Is there anyway to save old logging or get to it some other way? Sorry, but I am still learning how to work with Docker so I don't know these things. (I have tried using "until" in docker logs.)

Just as before. If we just stop and start IoT Edge after the failures the publisher and the EdgeHub logs that they are sending telemetry to the IoT Hub OK. However nothing ever reaches the IoT Hub. We need to reinstall to make it work again.

Jan-Olof, 2018-03-27:
Update: Windows Server 2016 running OK for 13 days. Ubuntu server running OK for 6 days. Windows 10 stopped working again two days ago.

So, I guess the problem is either the Windows 10 machine or its network environment. However since everything else (except sending telemetry using IoT Edge) continue to work I still think it would be important to find out the problem. However since this is not our production environment we will not investigate further. Instead we have decided to use an Ubuntu machine to run these tests in the production environment.

If you want more information I will be happy to help. Just tell me what to do.

varunpuranik, 2018-03-27:
Thanks for the update. 
I am glad to know that it seems to be running fine on Ubuntu.
As for the Windows 10, I am not aware of any differences between Win10 and Win Server 2016. What kind of errors are you seeing on Windows 10, or does it simply stop working? Can you provide the EdgeHub logs?

Jan-Olof, 2018-03-28:
We don't see any errors. It just stops working again. It happened this morning. Last message sent at 09:33 UTC. Here are logs from publisher and EdgeHub:

[publisher.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1855586/publisher.log.txt)

[edgehub.log.txt](https://github.com/Azure/iot-edge-opc-publisher/files/1855594/edgehub.log.txt)

Jan-Olof, 2018-03-28:
Update to last post: We may have found the reason to this problem. When looking in the OPC UA Server we found that it no longer has any active tags listed on its connected client. (Unlike in the publisher log.) So we reinitialized the OPC UA Server and after a little while the messages to the IoT Hub started again. 

So, maybe this has been the problem all along but we didn't realize it since we only looked in the IoT Edge logs? (The publisher says that the OPC UA connection is OK.) At least it seems to have been the problem this time.

varunpuranik, 2018-03-28:
Thanks for the update Jan-Olof. Glad to know you were able to resolve your issue. I will go ahead and close this issue. If you run into any more issues, please re-open the issue, or file a new one. 
Thanks.

Jan-Olof, 2018-03-29:
After 14 days the other test machine running Windows Server 2016 also stopped working. We were able to resolve it the same way, so it looks like we have narrowed down the problem. I did find a few more things in the EdgeHub log though. If it is of any interest I have pasted it below.

 I agree that you can close the issue if you want, since we can't pin the problem to the publisher (or even IoT Edge).

```
2018-03-28 17:01:37.722 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - ConnectionLost for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:01:37.799 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - ConnectionEstablished for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:01:37.800 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStateStoragePersistenceProvider] - Cloud connection established, setting subscriptions for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:01:37.800 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStatePersistenceProvider] - Adding subscription $iothub/twin/res/# for client ikea-msf-edge/iot-edge-opc-publisher.
2018-03-28 17:01:37.800 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStateStoragePersistenceProvider] - Set subscriptions from session state for ikea-msf-edge/iot-edge-opc-publisher on cloud reconnect
2018-03-28 17:01:37.800 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Processing ConnectionEstablished for device ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:01:37.801 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Getting twin for ikea-msf-edge/iot-edge-opc-publisher on ConnectionEstablished
2018-03-28 17:01:37.807 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Getting twin for device ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:01:37.808 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Successfully got twin for ikea-msf-edge/iot-edge-opc-publisher from cloud at desired version 1 reported version 1
2018-03-28 17:01:37.808 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Local twin for ikea-msf-edge/iot-edge-opc-publisher at higher or equal desired version 1 compared to cloud 1 or reported version 1 compared to cloud 1
2018-03-28 17:04:07.726 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - ConnectionLost for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:04:07.786 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - ConnectionEstablished for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:04:07.786 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStateStoragePersistenceProvider] - Cloud connection established, setting subscriptions for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:04:07.786 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Processing ConnectionEstablished for device ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:04:07.787 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Getting twin for ikea-msf-edge/iot-edge-opc-publisher on ConnectionEstablished
2018-03-28 17:04:07.787 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStatePersistenceProvider] - Adding subscription $iothub/twin/res/# for client ikea-msf-edge/iot-edge-opc-publisher.
2018-03-28 17:04:07.787 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStateStoragePersistenceProvider] - Set subscriptions from session state for ikea-msf-edge/iot-edge-opc-publisher on cloud reconnect
2018-03-28 17:04:07.788 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - ConnectionLost for ikea-msf-edge/iot-edge-opc-publisher
2018-03-28 17:04:07.790 +00:00 [WRN] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Error in connection established callback for client ikea-msf-edge/iot-edge-opc-publisher - System.Collections.Generic.KeyNotFoundException The given key was not present in the dictionary.
```

gasherma, 2018-06-07:
Per comment from Jan-Olof - closing issue as not clearly attributable to the publisher

closed 2018-06-07 by gasherma