Azure/iot-edge-opc-publisher-v1 #112: Raspberry Pi 3 "exec format error"

Eichi87, 2018-04-18:
I was wondering if it is possible to implement opc publisher in a Raspberry Pi 3.
I'm running stock Raspbian and using openAAS with acplt/rte as OPC UA server.
Docker installation worked just fine but during the build process of iot-edge-opc-publisher i'm running in an error.

During "Step 4/7: RUN dotnet restore ---> Running in b7cc0f1fc934" pops a red error message: "standard_init_linux.go:190: exec user process caused "exec format error""
I think this may be caused by the ARM architecture. Am I right with this? And if yes is there a workaround available for this? Or maybe I'm using the wrong Dockerfile?
Here's what I did:
`~ $ git clone https://github.com/Azure/iot-edge-opc-publisher`
`~ $ cd iot-edge-opc-publisher/`
`~/iot-edge-opc-publisher $ docker build -f Dockerfile -t iot-edge-opc-publisher .`
And as mentioned above during Step 4 it stops with the given error message.

I would be grateful if someone could help me or at least tell me if it is even possible to run opc publisher on a Raspberry Pi 3.

FluxReality, 2018-04-19:
Have the same issue.

I think "someone" have to redesign the docker file to use multi-stage build so that the ARM runtime of dotnet is being used when deploying. (Like [this]( https://github.com/dotnet/dotnet-docker-samples/tree/master/dotnetapp-dev) example)

At least that's what I'm trying at the moment.

Eichi87, 2018-04-19:
I've never worked with docker files before so that's a problem for me.
Still I will have a look at it and in case there is no current solution and I can resolve the problem by myself I definitely will get back to this topic.
Would you share your findings with me and the rest of the world when you can resolve the issue?

FluxReality, 2018-04-20:
Ok, so I got it working, by changing the **Dockerfile** to this:
```
FROM microsoft/dotnet:2.0-sdk AS build

COPY /src /build/src

WORKDIR /build/src
RUN dotnet restore
RUN dotnet publish --configuration Release --output /build/out

FROM microsoft/dotnet:2.0-runtime-stretch-arm32v7 AS runtime
WORKDIR /docker
COPY --from=build /build/out ./
ENTRYPOINT ["dotnet", "OpcPublisher.dll"]
```
Probably a good idea to create targeted builds for the various architectures, or at least separate dockerfiles.

Eichi87, 2018-04-20:
That was really fast :smiley:

Unfortunately this doesn't work for me:
`Step 1/9 : FROM microsoft/dotnet:2.0-sdk AS build
2.0-sdk: Pulling from microsoft/dotnet
no matching manifest for linux/arm in the manifest list entries`
Do you run raspbian or any other Linux distribution?
If I change `AS build` back to `-jessie` i run in the same error as before.
Did you change anything else or did you do anything special before running the docker file?
Maybe I need to build this on a 64bit machine and copy over to the PI?

I very much appreciate your help and am very thankful! :smile:

FluxReality, 2018-04-20:
My device is a Raspberry PI 3 running Raspbian GNU/Linux 9 (stretch).

But I am building everything on my windows development computer. I don't think the SDK for ARM has been released yet (see this [issue](https://github.com/dotnet/announcements/issues/29))

sebader, 2018-06-06:
.NET Core 2.1 just got released a few days ago and also brings now ready made [docker containers for ARM](https://hub.docker.com/r/microsoft/dotnet/ ). Using this, I successfully built and ran the OPC Publisher module on a Raspberry PI 3 today :)
I just changed the base image in the Dockerfile to 2.1-sdk-stretch-arm32v7 and it built (just took a while on the raspi itself...)
I am planning to do a PR on this when I find the time.

sebader, 2018-06-08:
@FluxReality my PR is out. You can have a look a the Dockerfile for arm32v7 already and see if that works for you

FluxReality, 2018-08-10:
@sebader I had to make some modifications to the arm32v7 file as it stands, to get it to compile on my windows machine before deploying to my rasppi device as an IoT Edge Module. I changed the following line:

**`ARG build_base_tag=2.1-sdk-bionic-arm32v7`**

to

**`ARG build_base_tag=2.1-sdk-alpine`**

In the docker-file keeping the rest as it was. Then all ting builds and starts (except an apparent other issue #145) when trying to run it as a module on the GA release of Azure Iot Edge.