Azure/iot-edge-opc-publisher-v1 #114: OPC UA PubSub 1.04 Compliance

mpostol, 2018-04-21:
The Readme.md file states:

> This reference implementation demonstrates how Azure IoT Edge can be used to connect to existing OPC UA servers and publishes JSON encoded telemetry data from these servers in OPC UA "Pub/Sub" format (using a JSON payload) to Azure IoT Hub. 

It requires that the message created by:

`IotHubMessaging.CreateJsonMessageAsync(OpcPublisher.OpcMonitoredItem.MessageData)`

has to be compliant with the specification:

- [OPC Unified Architecture Specification Part 14: PubSub Release 1.04 February 06, 2018][OPC.UA.PubSub]

My concern is what is the reason to create the message following this specification. As far as I understand this encoding. i.e. the JSON based message mapping allows OPC UA Applications to interoperate with web and enterprise software that use this format. Is there any plan to:
- make the Azure IoT Edge OPC UA PubSub JSON encoding aware
- compliant with OPC UA PubSub Application and offer the Subscriber role.

[OPC.UA.PubSub]: https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-14-pubsub/

ggokka, 2018-10-29:
@mpostol Have you got an answer about OPC UA PubSub 1.04 Compliance?

mpostol, 2018-11-02:
@ggokka I have not got any response. It seems that one reason is stability of the Part 14. there are a lot of issues reported:
https://opcfoundation-onlineapplications.org/mantis/view_all_bug_page.php
There is a discussion you may be interested:
https://lnkd.in/gAtSxaX

mpostol, 2019-02-10:
Yesterday I have got official information from OPCF:

> With respect to IPR 2.1, clause 2 we inform you that the following amendment for version 1.04 of the OPC UA specification has been published
>
> -  **Amendment 6 UADP Header Layouts**
>
> The document is available for members-only at this location: [Errata and Amendments](https://opcfoundation.org/developer-tools/specifications-unified-architecture/errata-and-amendments/)
> The review period to opt out claims starts today 09.02.2019 and will end in 90 days.

Unfortunately, the document is not available for me due to membership limitations.   It seems that any activity related to PubSub must be postponed until the spec will be stable. 

My recommendation is: label it as waiting.