Azure/iot-edge-opc-publisher-v1 #115: JSON message format to IotHub

davebanthorpe, 2018-04-30:
Hi - I have the module up and running pulling data from a BACnet device (via Kepware). The format of the output JSON is as follows:

{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_1.Water Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "DisplayName": "Water Temperature",
  "Value": 37.1,
  "SourceTimestamp": "2018-04-30T14:09:59.4983294Z"
},{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_0.Indoor Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "DisplayName": "Indoor Temperature",
  "Value": 20.9,
  "SourceTimestamp": "2018-04-30T14:10:00.4993239Z"
},
........

How do I configure it to send it as the following format (i.e. merging the DisplayName and Value attributes into a single parameter) so that I can via the streaming data in PowerBI?

{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_1.Water Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "Water Temperature": "37.1",
  "SourceTimestamp": "2018-04-30T14:09:59.4983294Z"
},{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_0.Indoor Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "Indoor Temperature": "20.9",
  "SourceTimestamp": "2018-04-30T14:10:00.4993239Z"
},
.......

davebanthorpe, 2018-05-11:
which command option is that so I can test?

hansgschossmann, 2018-06-05:
Hi Dave, sorry i was on vacation for a while.

In my above mentioned fork the switch is --ic and you need to pass in the IoTCentral connection string via --dc. 

Here is the inital change if you want to take it over into your code: https://github.com/hansgschossmann/iot-edge-opc-publisher/commit/9cbe3c469561fc0f73f3167cebf2612af0ac77f5

I will merge these changes into this repository later in June.

Bodera, 2019-03-20:
Hi people, that's an old issue, but I'm trying to achieve the same goal that the @davebanthorpe, and I also dealing with multiple OPC-UA nodes. If it's preferable to open a new issue, I will gladly do it.
By passing the parameter `--ic` in the OPC Publisher I got the results as expected:
```json
{"myOpcTagName": "myNumericValueAsString"},
{"otherOpcTagName": "otherNumericValueAsString"}
```
In order to deal with that adversity I've created a JavaScript UDF for the ASA job presented as below:
```javascript
function main(input) {
   return Number.isNaN(parseFloat(input.myOpcTagName)) ? 0 : parseFloat(input.myOpcTagName);
}
```
The use of ternary operators gives to us the value of the node if the name of JSON object has matched and 0 if it doesn't. Works fine in the dashboards of Power BI Service. 
But I'm more ambicious and want to retain the previous value of my node instead of that 0. I look up for aggregate functions using JavaScript UDA but the telemetry sents a JSON file with a nested object called `IoTHub` as illustrated:
```json
{
      "RandomDouble":"-10.505183000000002",
      "EventProcessedUtcTime":"2019-03-20T16:21:14.6071698Z",
      "PartitionId":0,
      "EventEnqueuedUtcTime":"2019-03-20T16:21:05.1550000Z",
      "IoTHub":{
         "MessageId":null,
         "CorrelationId":null,
         "ConnectionDeviceId":"opcuazure-iot-central",
         "ConnectionDeviceGenerationId":"636886956385548859",
         "EnqueuedTime":"2019-03-20T16:21:05.1430000Z",
         "StreamId":null
      }
   }
```
Can you people give me some orientation about how to procede?
Thanks in advance, I'm learning a lot with the OPC Publisher tool.