Azure/iot-edge-opc-publisher-v1 #116: OPC Publisher won't send messages again after a lost connection

tbuquet, 2018-05-05:
Hello,

I'm currently having an issue with OPC Publisher. If for any reason the connection is lost between OPC Publisher and a OPC Server, it will attempt to reconnect (and seems to do so successfully), but then OPC Publisher won't receive node updates anymore and won't send anything back to IoT Hub.

Here's the diagnostic:
`5/5/2018 00:15:32.470 ==========================================================================
5/5/2018 00:16:32.479 ==========================================================================
5/5/2018 00:16:32.482 OpcPublisher status @ 5/5/2018 5:16:32 AM (started @ 5/4/2018 8:02:08 PM)
5/5/2018 00:16:32.484 ---------------------------------
5/5/2018 00:16:32.486 OPC sessions: 2
5/5/2018 00:16:32.487 connected OPC sessions: 2
5/5/2018 00:16:32.489 connected OPC subscriptions: 2
5/5/2018 00:16:32.490 OPC monitored items: 4466
5/5/2018 00:16:32.492 ---------------------------------
5/5/2018 00:16:32.494 monitored items queue bounded capacity: 8192
5/5/2018 00:16:32.495 monitored items queue current items: 0
5/5/2018 00:16:32.497 monitored item notifications enqueued: 83489
5/5/2018 00:16:32.499 monitored item notifications enqueue failure: 0
5/5/2018 00:16:32.500 monitored item notifications dequeued: 83489
5/5/2018 00:16:32.502 ---------------------------------
5/5/2018 00:16:32.503 messages sent to IoTHub: 4978
5/5/2018 00:16:32.504 last successful msg sent @: 5/5/2018 4:22:28 AM
5/5/2018 00:16:32.506 bytes sent to IoTHub: 19684740
5/5/2018 00:16:32.507 avg msg size: 3954
5/5/2018 00:16:32.509 msg send failures: 0
5/5/2018 00:16:32.511 messages too large to sent to IoTHub: 0
5/5/2018 00:16:32.512 times we missed send interval: 167
5/5/2018 00:16:32.514 ---------------------------------
5/5/2018 00:16:32.516 current working set in MB: 89
5/5/2018 00:16:32.518 --si setting: 5
5/5/2018 00:16:32.519 --ms setting: 262144
5/5/2018 00:16:32.521 --ih setting: Amqp_Tcp_Only
5/5/2018 00:16:32.523 ==========================================================================
5/5/2018 00:16:54.241 SECURE CHANNEL RENEWED [.NetStandard ServerChannel UA-TCP 1.03.351UA-TCP 1.03.351] [ID=5]
5/5/2018 00:17:25.580 Session endpoint: opc.tcp://x.x.x.x:4096/iaopcua/Basic128Rsa15 has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.580 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.604 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.618 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.622 Session endpoint: opc.tcp://x.x.x.x:4096/iaopcua/Basic128Rsa15 has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.703 ***EXCEPTION*** BadNoSubscription
5/5/2018 00:17:25.998 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.041 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.055 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.080 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.103 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.124 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.141 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.157 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.167 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.213 SessionId: ns=1;g=bc5cf924-1457-4cc2-b0b2-45f314acfde4
5/5/2018 00:17:26.181 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.224 SessionId: ns=1;i=3
5/5/2018 00:17:26.228 Missed KeepAlives: 1
5/5/2018 00:17:26.204 SessionId: ns=1;g=bc5cf924-1457-4cc2-b0b2-45f314acfde4
5/5/2018 00:17:26.190 SessionId: ns=1;i=3
5/5/2018 00:17:26.203 SessionId: ns=1;g=bc5cf924-1457-4cc2-b0b2-45f314acfde4
5/5/2018 00:17:26.220 Missed KeepAlives: 1
5/5/2018 00:17:26.241 Session endpoint: opc.tcp://10.4.13.63:4096/iaopcua/Basic128Rsa15 got a keep alive after 1 was missed.
5/5/2018 00:17:26.248 Missed KeepAlives: 2
5/5/2018 00:17:26.257 Missed KeepAlives: 2
5/5/2018 00:17:26.277 Session endpoint: opc.tcp://x.x.x.x:49320/ got a keep alive after 3 were missed.
5/5/2018 00:17:26.294 Session endpoint: opc.tcp://x.x.x.x:49320/ got a keep alive after 3 were missed.
5/5/2018 00:17:26.267 Missed KeepAlives: 3
5/5/2018 00:17:26.293 ***EXCEPTION*** BadNoSubscription
5/5/2018 00:17:32.513 ==========================================================================
5/5/2018 00:17:32.517 OpcPublisher status @ 5/5/2018 5:17:32 AM (started @ 5/4/2018 8:02:08 PM)
5/5/2018 00:17:32.518 ---------------------------------
5/5/2018 00:17:32.520 OPC sessions: 2
5/5/2018 00:17:32.521 connected OPC sessions: 2
5/5/2018 00:17:32.523 connected OPC subscriptions: 2
5/5/2018 00:17:32.525 OPC monitored items: 4466
5/5/2018 00:17:32.526 ---------------------------------
5/5/2018 00:17:32.528 monitored items queue bounded capacity: 8192
5/5/2018 00:17:32.530 monitored items queue current items: 0
5/5/2018 00:17:32.531 monitored item notifications enqueued: 83489
5/5/2018 00:17:32.533 monitored item notifications enqueue failure: 0
5/5/2018 00:17:32.534 monitored item notifications dequeued: 83489
5/5/2018 00:17:32.536 ---------------------------------
5/5/2018 00:17:32.537 messages sent to IoTHub: 4978
5/5/2018 00:17:32.540 last successful msg sent @: 5/5/2018 4:22:28 AM
5/5/2018 00:17:32.542 bytes sent to IoTHub: 19684740
5/5/2018 00:17:32.544 avg msg size: 3954
5/5/2018 00:17:32.545 msg send failures: 0
5/5/2018 00:17:32.547 messages too large to sent to IoTHub: 0
5/5/2018 00:17:32.549 times we missed send interval: 168
5/5/2018 00:17:32.550 ---------------------------------
5/5/2018 00:17:32.553 current working set in MB: 89
5/5/2018 00:17:32.555 --si setting: 5
5/5/2018 00:17:32.557 --ms setting: 262144
5/5/2018 00:17:32.559 --ih setting: Amqp_Tcp_Only
5/5/2018 00:17:32.560 ==========================================================================
5/5/2018 00:18:32.569 ==========================================================================
5/5/2018 00:18:32.576 OpcPublisher status @ 5/5/2018 5:18:32 AM (started @ 5/4/2018 8:02:08 PM)
5/5/2018 00:18:32.578 ---------------------------------
5/5/2018 00:18:32.580 OPC sessions: 2
5/5/2018 00:18:32.582 connected OPC sessions: 2
5/5/2018 00:18:32.584 connected OPC subscriptions: 2
5/5/2018 00:18:32.586 OPC monitored items: 4466
5/5/2018 00:18:32.588 ---------------------------------
5/5/2018 00:18:32.589 monitored items queue bounded capacity: 8192
5/5/2018 00:18:32.591 monitored items queue current items: 0
5/5/2018 00:18:32.593 monitored item notifications enqueued: 83489
5/5/2018 00:18:32.595 monitored item notifications enqueue failure: 0
5/5/2018 00:18:32.596 monitored item notifications dequeued: 83489
5/5/2018 00:18:32.598 ---------------------------------
5/5/2018 00:18:32.599 messages sent to IoTHub: 4978
5/5/2018 00:18:32.601 last successful msg sent @: 5/5/2018 4:22:28 AM
5/5/2018 00:18:32.603 bytes sent to IoTHub: 19684740
5/5/2018 00:18:32.604 avg msg size: 3954
5/5/2018 00:18:32.606 msg send failures: 0
5/5/2018 00:18:32.607 messages too large to sent to IoTHub: 0
5/5/2018 00:18:32.609 times we missed send interval: 168
5/5/2018 00:18:32.611 ---------------------------------
5/5/2018 00:18:32.613 current working set in MB: 90
5/5/2018 00:18:32.615 --si setting: 5
5/5/2018 00:18:32.617 --ms setting: 262144
5/5/2018 00:18:32.619 --ih setting: Amqp_Tcp_Only
5/5/2018 00:18:32.621 ==========================================================================`

I'm not entirely sure what's going on, it seems to get back the keepalive messages after a connection was dropped, 5 being the max amount of keepalive to send before the session is disconnected. Any idea what might be causing this issue?

Thanks!

hansgschossmann, 2018-05-06:
Are you on master or on maintenance_2_1? Master is using a new stack which you may try. If you are already on master we need to analyze further.

Hans

Jan-Olof, 2018-05-07:
I had the same kind of issue last week. The OPC UA Server stopped working and when it got back up one of the nodes was not working. This caused an exception in the publisher. Still the publisher seemed to reconnect, but no data was sent to the hub. Restarting the IoT Edge just resulted in the logs saying that data was sent to the hub but nothing arrived. I had to reinstall IoT Edge to get the publisher to work again.

I should be able to recreate this so if you need more logs etc let me know and I will give it a try. (I am using the master.)

hansgschossmann, 2018-05-07:
It would be great if you could send me some logs.

Jan-Olof, 2018-05-07:
Sorry, can't recreate the problem now in our test environment. Everything is reconnecting and sending as it should despite exceptions. Maybe somethings else was the root to the problems last week, or maybe we were just running an older version at that time without knowing it. 

Let's hope it doesn't happen again. If it does I will let you know.

tbuquet, 2018-05-08:
Hello,

Thank you for your answers.

I am currently trying to reproduce the issue using the issue in a local environment with the OPC demo servers from OPC Foundation, without success so far. I will update this ticket when I can reproduce.

It might be a coincidence, but I haven't experienced a disconnection yet (about 2 days) since I lowered the amount of nodes being monitored.

tbuquet, 2018-05-08:
Hello,

I haven't been able to reproduce the issue yet. 
More specifically, I am troubled by the fact that the keep alive trace is out of order:
5/5/2018 00:17:25.580 Session endpoint: opc.tcp://x.x.x.x:4096/iaopcua/Basic128Rsa15 has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.580 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.604 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.618 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.622 Session endpoint: opc.tcp://x.x.x.x:4096/iaopcua/Basic128Rsa15 has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.703 EXCEPTION BadNoSubscription
5/5/2018 00:17:25.998 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.041 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.055 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.080 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.103 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.124 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.141 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.157 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.167 Good publish requests: 0, KeepAlive interval: 2000

OPC Publisher is running as a edge module in a VM, could this be the result of a nested environment?

hansgschossmann, 2018-06-05:
@YohskDista: you only will receive data when the connection to the OPC server is successfull. Does this repro consistently in your environment? Which OPC server are you using?

YohskDista, 2018-06-06:
For now I am trying to do this in local on my computer. The next step will be to have a central server that regroups all the OPC Servers connections.
I am using the OPC Traeger solution for my test. You can download the nuget package called Opc.UaFx.Advanced v2.2.1 in the Nuget package manager of Visual Studio to make some tests.

YohskDista, 2018-06-19:
After a bit more research, it appears that the issue lies in the fact that the `OpcMonitoredItem.Notification` event is not triggered in some edge cases. 

For us, it happens sometimes after the server restarts. We can see in the logs that the connection was correctly re-established to the server, that the items are correctly getting monitored, but adding a breakpoint into `OpcMonitoredItem.MonitoredItem_Notification`, we can see that the event is never triggered, and we don't receive updates from the OPC server.

Here's the output of the last elements of the logs, starting from a deconnection that we did on purpose (stopped the OPC server).

```
19.06.18 08:31:47.498 Session creation to endpoint 'opc.tcp://localhost:8995/' failed 1 time(s). Please verify if server is up and Publisher configuration is correct. BadNotConnected 'Error establishing a connection: BadNotConnected'

========================================
Id: BadNotConnected
Description: Error establishing a connection: BadNotConnected
>>> Error establishing a connection: BadNotConnected
---    at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
---    at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
---    at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints)
---    at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris)
---    at Opc.Ua.Client.CoreClientUtils.SelectEndpoint(String discoveryUrl, Boolean useSecurity, Int32 operationTimeout)
---    at OpcPublisher.OpcSession.<ConnectSessionAsync>d__24.MoveNext() in C:\ProjDotNet\iot-edge-opc-publisher\src\OpcSession.cs:line 512
===
Id: BadNotConnected
Description: BadNotConnected
>>> BadNotConnected
---    at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
---    at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.OnConnectOnDemandComplete(Object state)
========================================

19.06.18 08:31:57.525 Connect and monitor session and nodes on endpoint 'opc.tcp://localhost:8995/'.
19.06.18 08:31:57.531 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.03.352UA-TCP 1.03.352] [ID=3] Connected To: opc.tcp://localhost:8995/ [None/None/Binary]
19.06.18 08:31:57.549 Create secured session for endpoint URI 'opc.tcp://localhost:8995/' with timeout of 20000 ms.
19.06.18 08:31:57.553 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.03.352UA-TCP 1.03.352] [ID=5] Connected To: opc.tcp://localhost:8995/ [None/None/Binary]
19.06.18 08:31:57.581 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.03.352UA-TCP 1.03.352] [ID=7] Connected To: opc.tcp://localhost:8995/ [None/None/Binary]
19.06.18 08:31:57.725 Session successfully created with Id ns=4;i=1522206136.
19.06.18 08:31:57.768 The session to endpoint 'opc.tcp://localhost:8995/' has 5 entries in its namespace array:
19.06.18 08:31:57.770 Namespace index 0: http://opcfoundation.org/UA/
19.06.18 08:31:57.771 Namespace index 1: urn:###:###
19.06.18 08:31:57.772 Namespace index 2: http://###/###
19.06.18 08:31:57.772 Namespace index 3: http://###/###
19.06.18 08:31:57.773 Namespace index 4: http://opcfoundation.org/UA/Diagnostics
19.06.18 08:31:57.776 The server on endpoint 'opc.tcp://localhost:8995/' supports a minimal sampling interval of 0 ms.
19.06.18 08:31:57.800 Created subscription with id 1 on endpoint 'opc.tcp://localhost:8995/'
19.06.18 08:31:57.802 Create subscription on endpoint 'opc.tcp://localhost:8995/' requested OPC publishing interval is 0 ms. (revised: 0 ms)
19.06.18 08:31:57.803 Start monitoring items on endpoint 'opc.tcp://localhost:8995/'. Currently monitoring 0 items.
19.06.18 08:31:57.830 Created monitored item for node 'ns=2;s=Metadata/AppState' in subscription with id '1' on endpoint 'opc.tcp://localhost:8995/'
19.06.18 08:31:57.832 Now monitoring 1 items in subscription with id '1'
19.06.18 08:31:57.835 Done processing unmonitored items on endpoint 'opc.tcp://localhost:8995/'. Now monitoring 1 items in subscription with id '1'.
```

DaveDashFTW, 2018-11-19:
I want to add that I am running this across multiple sites across the globe, and I notice the same problem.

As soon as the *subscription* is lost, the OPC publisher doesn't bother to re-send data.

I am using OPC Server (typically OPC-DA) -> KEPServerEX -> OPC Publisher. 

Without fail, whenever "BadNoSubscription" is logged in OPC-Publisher, data fails to send after reconnect:

11/16/2018 23:56:41.584 ***EXCEPTION*** BadRequestTimeout
11/16/2018 23:56:41.989 PUBLISH #8006863 - Unhandled error during Publish. BadRequestTimeout 'BadRequestTimeout'

========================================
Id: BadRequestTimeout
Description: BadRequestTimeout
>>> BadRequestTimeout
---    at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
---    at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
---    at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
---    at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
========================================

**11/16/2018 23:56:42.046 ***EXCEPTION*** BadNoSubscription**
11/17/2018 00:17:10.751 SECURE CHANNEL RENEWED [.NetStandard ServerChannel UA-TCP 1.03.351UA-TCP 1.03.351] [ID=3280577939]

I am currently running OPC-Publisher as a guest executable in Service Fabric and I am writing a custom service that monitors the logs and fires a restart IoT Edge when it detects BadNoSubscription, but for large number of tags constantly restarting IoT Edge isn't ideal.