Azure/iot-edge-opc-publisher-v1 #118: Using username/password instead of certificate to get authenticated to an OPC UA server

kotselio, 2018-05-16:
Hello! It's often the case that an OPC UA server does not have a certificate, but only username/password type of credentials. AFAIU there is no support for this feature in the current version of the publisher. Any chance that it's gonna be added?

tbuquet, 2018-05-17:
Hello, I've had a similar issue and can confirm that it isn't implemented. However implementing it isn't too difficult:
Line 523 in the OpcSession.cs file:
`OpcUaClientSession = await Session.Create(
                            PublisherOpcApplicationConfiguration,
                            configuredEndpoint,
                            true,
                            false,
                            PublisherOpcApplicationConfiguration.ApplicationName,
                            timeout,
                            new UserIdentity(new AnonymousIdentityToken()),
                            null);`

to something like:
`OpcUaClientSession = await Session.Create(
                            PublisherOpcApplicationConfiguration,
                            configuredEndpoint,
                            true,
                            false,
                            PublisherOpcApplicationConfiguration.ApplicationName,
                            timeout,
                            _useLogin ? new UserIdentity(_login, _password) : new UserIdentity(new AnonymousIdentityToken()),
                            null);
`

mm-mhr, 2018-05-24:
I tried as proposed by tbuquet.

I get the following exception:
Id: BadSecurityChecksFailed
Description: BadSecurityChecksFailed
>>> BadSecurityChecksFailed
---    at Opc.Ua.Client.Session.<Create>d__73.MoveNext()
--- --- End of stack trace from previous location where exception was thrown ---
---    at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
---    at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
---    at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
---    at OpcPublisher.OpcSession.<ConnectSessionAsync>d__22.MoveNext() in /home/mm/Source/EdgeModules/opcuapublisher/src/OpcSession.cs:line 531

>>> BadSecurityChecksFailed
---    at Opc.Ua.CertificateValidator.<InternalValidate>d__22.MoveNext()
===
Id: BadSecurityChecksFailed
Description: BadSecurityChecksFailed
>>> BadSecurityChecksFailed
---    at Opc.Ua.CertificateValidator.<InternalValidate>d__22.MoveNext()

In tcpdump I don't see any action for this, so I guess the exception is thrown before the actual session is opened.
Any Idea?

hansgschossmann, 2018-06-05:
Sorry for the late response.

You are right that is not implemented. I have not yet a good idea where to keep the credentials in a secure place, since those would be required to be set and persisted per session (per OPC UA server).

Ideas?

mm-mhr, 2018-06-06:
I agree that having the credentials in the config file is from security point of view a no go.
However, to have something working it should be acceptable, at least during development.

In general I think that usage of credentials in a config file / the device twin  is a typical use case, not just for the OPC UA publisher. So it would be nice if there was a general solution provided as part of Azure IoT edge.

anhame, 2019-03-04:
Anyhow it would be great if the UserName/Password properties within PublishNodesMethodRequestModel could be removed as long as this feature isn't going to be impemented.  One could think that this might work when providing according information (as I did).  ;-)