Azure/iot-edge-opc-publisher-v1 #119: Message batching via JSON array

jcharlytown, 2018-05-29:
Hi,

we're currently evaluating whether or not it would be feasible to connect multiple OPC UA endpoints (and therefore multiple logical or even physical devices) via a single instance of the opc publisher. Doing so, we ran into the fact that the opc publisher batches messages "manually" using a JSON array of messages - on application layer if you will. In our scenario, this means we might end up with messages from different OPC UA endpoints within the same IoT Hub message. This in turn becomes challenging later on within the processing pipeline, e.g., when joining incoming messages with state from CosmosDB via Azure Functions - we would need to join with multiple documents within a single trigger.

My question is the following: Is this choice intentional? Is there a reason for this design we're not seeing? It seems that the IoT Hub c# SDK offers [DeviceClient.SendEventBatchAsync](https://docs.microsoft.com/en-us/dotnet/api/microsoft.azure.devices.client.deviceclient.sendeventbatchasync?view=azure-dotnet#Microsoft_Azure_Devices_Client_DeviceClient_SendEventBatchAsync_System_Collections_Generic_IEnumerable_Microsoft_Azure_Devices_Client_Message__) for transport layer optimization, which would potentially avoid this issue.

Also, any idea for a workaround, e.g., disabling batching or splitting messages up within Azure?

jcharlytown, 2018-06-05:
Any comments on this?

hansgschossmann, 2018-06-05:
Sorry for the late response.

The batching was intentional to save cost and done at "app level" since not all transports support it: https://github.com/Azure/azure-iot-sdk-csharp/issues/121

As a workaround i suggest to set the message size parameter and the send interval to 0 via the --ms and --si options.

The batch is sent as a JSON array, so you can split it on the cloud side, but this requires some code on your side.

jcharlytown, 2018-06-05:
Thanks for your response! This is valuable insight.

Microsoft advertises this project as part of the reference architecture for Azure IoT. You suggested that the JSON array should be split up on the cloud side. I'd like to discuss how to achieve this best, however, I am not entirely sure that this is the right place to do so. If it's not, can you refer me to a better place?

We evaluated both Azure Stream Analytics (ASA) as well as Azure Functions for the purpose of splitting the array up. For an ASA job, it seems that multiple output "rows" gets batched again into a single Event Hub message again: https://stackoverflow.com/questions/42213418/stream-analytics-to-event-hub-unexpectedly-concatenating-events

Regarding Azure Functions, it seems that output bindings also imply one output event per trigger, i.e., a JSON array: https://docs.microsoft.com/en-us/azure/azure-functions/functions-bindings-event-hubs#output---javascript-example

Any further pointers?