Azure/iot-edge-opc-publisher-v1 #121: IoT Edge on MacOS

remcoploeg, 2018-06-11:
Hi,

I have setup an IoT Edge on MacOS and the samples with tempSensor and ASA jobs are running fine. Now I want to test with the IoT Edge module for OPC-UA and a OPC UA Simulation server to get data from that server and send it to the IoT Hub. I have configured the IoT Edge OPC-Publisher module on the IoT Edge. It's running, but have some questions around it: 

1.  Error
```
Access to the path '/docker/publishednodes.json' is denied.
06/11/2018 19:31:36:    at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirectory, Func`2 errorRewriter)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at System.IO.File.AsyncStreamWriter(String path, Encoding encoding, Boolean append)
   at System.IO.File.WriteAllTextAsync(String path, String contents, Encoding encoding, CancellationToken cancellationToken)
   at OpcPublisher.PublisherNodeConfiguration.<UpdateNodeConfigurationFileAsync>d__20.MoveNext() in
```

Why is this?  I seems an win32 error?

2. I see on the log of the OPC_UC publisher IoT Edge module, that it's connected to my simulation server (I see that also on the simulation server)

```
connected OPC subscriptions: 2
06/11/2018 19:31:36: OPC monitored items: 4
06/11/2018 19:31:36: ---------------------------------
06/11/2018 19:31:36: monitored items queue bounded capacity: 8192
06/11/2018 19:31:36: monitored items queue current items: 0
06/11/2018 19:31:36: monitored item notifications enqueued: 0
06/11/2018 19:31:36: monitored item notifications enqueue failure: 0
06/11/2018 19:31:36: monitored item notifications dequeued: 0
06/11/2018 19:31:36: ---------------------------------
06/11/2018 19:31:36: messages sent to IoTHub: 0
06/11/2018 19:31:36: last successful msg sent @: 01/01/0001 00:00:00
06/11/2018 19:31:36: bytes sent to IoTHub: 0
06/11/2018 19:31:36: avg msg size: 0
06/11/2018 19:31:36: msg send failures: 0
06/11/2018 19:31:36: messages too large to sent to IoTHub: 0
06/11/2018 19:31:36: times we missed send interval: 0
06/11/2018 19:31:36: ---------------------------------
06/11/2018 19:31:36: current working set in MB: 85
06/11/2018 19:31:36: --si setting: 1
06/11/2018 19:31:36: --ms setting: 0
06/11/2018 19:31:36: --ih setting: Mqtt_Tcp_Only
```

But don't get any data. I have configured the published.json file like this:

```
[
    {
        // example for an EnpointUrl is: opc.tcp://win10iot:51210/UA/SampleServer
        "EndpointUrl": "opc.tcp://MacBook-Pro.localdomain:53530/OPCUA/SimulationServer",
        // Allows to access the endpoint with SecurityPolicy.None when set to 'false' (no signing and encryption applied to the OPC UA communication), default is true
        "UseSecurity": false,
        "OpcNodes": [
            // Publisher will request the server at EndpointUrl to sample the node with the OPC sampling interval specified on command line (or the default value: OPC publishing interval)
            // and the subscription will publish the node value with the OPC publishing interval specified on command line (or the default value: server revised publishing interval).
            {
                // The identifier specifies the NamespaceUri and the node identifier in XML notation as specified in Part 6 of the OPC UA specification in the XML Mapping section.
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=5"
            },
            // Publisher will request the server at EndpointUrl to sample the node with the OPC sampling interval specified on command line (or the default value: OPC publishing interval)
            // and the subscription will publish the node value with an OPC publishing interval of 4 seconds.
            // Publisher will use for each dinstinct publishing interval (of nodes on the same EndpointUrl) a separate subscription. All nodes without a publishing interval setting,
            // will be on the same subscription and the OPC UA stack will publish with the lowest sampling interval of a node.
            {
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=5",
                "OpcPublishingInterval": 4000
            },
            // Publisher will request the server at EndpointUrl to sample the node with the given sampling interval of 1 second
            // and the subscription will publish the node value with the OPC publishing interval specified on command line (or the default value: server revised interval).
            // If the OPC publishing interval is set to a lower value, Publisher will adjust the OPC publishing interval of the subscription to the OPC sampling interval value.
            {
                "ExpandedNodeId": "nsu=http://opcfoundation.org/UA/;i=5",
                // the OPC sampling interval to use for this node.
                "OpcSamplingInterval": 1000
            }
        ]
    },

    // the format below (NodeId format) is only supported for backward compatibility. you need to ensure that the
    // OPC UA server on the configured EndpointUrl has the namespaceindex you expect with your configuration.
    // please use the ExpandedNodeId format as in the examples above instead.
    {
        "EndpointUrl": "opc.tcp://MacBook-Pro.localdomain:53530/OPCUA/SimulationServer",
        "NodeId": {
            "Identifier": "ns=0;i=5"
        }
    }
    // please consult the OPC UA specification for details on how OPC monitored node sampling interval and OPC subscription publishing interval settings are handled by the OPC UA stack.
    // the publishing interval of the data to Azure IoTHub is controlled by the command line settings (or the default: publish data to IoTHub at least each 1 second).
]
```

But if I look at the logs of my simulation server I see a request on from the edge module:

```
ReadRequest
  RequestHeader=RequestHeader
    AuthenticationToken=i=4029648194
    Timestamp=2018-06-11 19:30:32.3314165 GMT
    RequestHandle=70
    ReturnDiagnostics=0
    AuditEntryId=null
    TimeoutHint=4000
    AdditionalHeader=null
  MaxAge=0.0
  TimestampsToReturn=TimestampsToReturn
    name=Neither
    ordinal=3
  NodesToRead=ReadValueId[1]
    [0]=ReadValueId
    NodeId=i=2259
    AttributeId=13
    IndexRange=null
    DataEncoding=null
```

I don't have in the configuration setup used 2259 as NodeID (but 5, like in the config above)

Any idea what can be wrong? I am connected, I think I have the right nodeID (5), but see in the simulation logs another nodeID used... maybe a simple question :)

sebader, 2018-06-12:
how did you mount the path where your publishednodes.json is stored? Can you post your IoT Edge deployment config?

remcoploeg, 2018-06-12:
Hi,

```
{
  "Hostname": "publisher",
  "Cmd": [
    "dotnet",
    "/build/out/OpcPublisher.dll",
    "publisher",
    "--pf",
    "/docker/publishednodes.json",
    "--lf",
    "/docker/publisher.log.txt",
    "--si",
    "1",
    "--ms",
    "0",
    "--di",
    "5",
    "--fd",
    "true",
    "--tm",
    "true",
    "--as",
    "true",
    "--vc",
    "true",
    "--ih",
    "Mqtt_Tcp_Only"
  ],
  "HostConfig": {
    "Binds": [
      "//tmp/docker/docker:/docker"
    ]
  }
}
```

sebader, 2018-06-12:
Can you remove the first / before /tmp? So using "/tmp/docker/docker"
Also, just to confirm: On your host, i.e. your Mac, the full path to the config is /temp/docker/docker/publishednodes.json?!

remcoploeg, 2018-06-12:
Yes, try this tonight (machine is not here). But I almost known for sure that /tmp/docker/docker is correct. Because if I change the connection string on the published nodes.json, I get a different error (and I see on the simulation server a connection)

sebader, 2018-06-12:
I have no idea about Docker on MacOS but is maybe the mount read-only (by default)? The module tries to about the json file (I'm not sure why, though). So if it does not have permission to do that, this could explain the error you are getting.

remcoploeg, 2018-06-12:
Ok thanks. Else I will try Windows, to see if the setup is working.

markusstuhler, 2019-04-30:
Hi @remcoploeg, does it work on windows or linux? If it does not work on windows or linux, please reopen.

closed 2019-04-30 by markusstuhler