Azure/iot-edge-opc-publisher-v1 #123: New release required

diegorapela, 2018-06-15:
Hi,
 
We are configuring the opc-publisher exclusively via its OPC server “façade”. This means, we do not have access to the “publishednodes.json” file. We also require that the messages sent to the iot-hub contains the expanded node id (nsu+id), and not the old format. Because of that, we depend on the fix pushed with the commit (“Fix async naming, fix usage of ExpandedNodeId syntax usage, fix locking”, https://github.com/Azure/iot-edge-opc-publisher/commit/6b6af1e7fbede015a37643d9abc7903375fda8ce). Unfortunately, this and many other (important) fixes came after the last release 2.1.4. Can you please create a new release that contains fixes pushed since 2.1.4?
 
Thanks,
Diego

jcharlytown, 2018-06-20:
Hi,

maybe I can help to clarify. First of all, we have a specific objective: We'd like the publisher to always use the ExpandedNodeId ("nsu=" syntax) when sending telemetry data to the IoT Hub, not plain NodeId ("ns=" syntax). Afaik, the publisher chooses the format with which the "MonitoredItem" has been configured in the first place: https://github.com/Azure/iot-edge-opc-publisher/blob/maintenance_2_1/src/OpcSession.cs#L172

What @diegorapela is basically saying is that we do not have access to the json file for configuring the nodes to publish, we need to do any configuration via the "PublishNode" and "UnpublishNode" OPC UA server methods. However, for version 2.1.4, "PublishNode" seems to not handle ExpandedNodeIds correctly. On input `NodeId="nsu=http://kuka.com/opcuaserver/;s=MotionDeviceSystem.MotionDevices.MotionDevice_1.Axes.Axis_1.ParameterSet.ActualPosition"` it ends up with:

```
"Result": [{
  "EndpointUrl": "opc.tcp://192.168.1.20:4840",
  "NodeId": {
    "Identifier": "s=nsu=http://kuka.com/opcuaserver/;s=MotionDeviceSystem.MotionDevices.MotionDevice_1.Axes.Axis_1.ParameterSet.ActualPosition"
  }
}]
```

On the other hand, using "latest" this issue seems to be fixed. As @diegorapela mentioned, it seems that https://github.com/Azure/iot-edge-opc-publisher/commit/6b6af1e7fbede015a37643d9abc7903375fda8ce is the commit in question. Hence our question: Would you consider this a bug fix which would be applicable for 2.1.X maintenance branch? If yes, would that then entail a 2.1.5 release?

hansgschossmann, 2018-06-20:
You are right, this was not handled correct. I will look into this.

Outlook for Android<https://aka.ms/ghei36> herunterladen

On the other hand, using "latest" this issue seems to be fixed. As @diegorapela<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fdiegorapela&data=02%7C01%7Cjohanng%40microsoft.com%7C990a3142ce244bae950d08d5d68d7855%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636650825465148293&sdata=7Si2Zyr8RwUQw8VccmHTiMWDsr9TqIgS5b59F4Gwev4%3D&reserved=0> mentioned, it seems that 6b6af1e<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FAzure%2Fiot-edge-opc-publisher%2Fcommit%2F6b6af1e7fbede015a37643d9abc7903375fda8ce&data=02%7C01%7Cjohanng%40microsoft.com%7C990a3142ce244bae950d08d5d68d7855%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636650825465148293&sdata=j4PnUwJtZ6WVRqoI%2Fl3p8xf6p7Wpp2pMBgOyTzjRAYA%3D&reserved=0> is the commit in question. Hence our question: Would you consider this a bug fix which would be applicable for 2.1.X maintenance branch? If yes, would that then entail a 2.1.5 release?

marcschier, 2019-07-02:
This should be in 2.4.x release and given that 2.x is backwards compatible we are not planning to touch 2.1.x releases, unless there is a good reason.   If latter, please re-open.

closed 2019-07-02 by marcschier